OpenDroneMap/ODM #469: Northing and Easting values in CSV file don't appear correct

piousgreek, 2017-02-08:
I may be misinterpreting this file, but the CSV file generated does not appear to contain complete point data.  I believe that in the northing and easting columns that it only contains the values for the seconds (not the minutes or degrees). 

[image]

The complet CSV is here:
https://www.dropbox.com/s/eu0xhzvejl24i03/odm_georeferenced_model.csv?dl=0

The geotagged source images can be found here:
https://www.dropbox.com/s/f1vvqos4l6cfxny/ShepherdDeWolf.zip?dl=0

The command I ran was:
python run.py --orthophoto-resolution 160.0 --force-focal 3.97 --pmvs-csize 1 --mesh-size 400000 --mesh-octree-depth 12 --project-path /Desktop/quickwork/geotagged

smathermather, 2017-02-08:
That looks like UTM, not geographic (lat, lon).

LucasMM86, 2017-02-08:
Yes, this file has UTM coordinates, not lat/lon. See https://en.wikipedia.org/wiki/Universal_Transverse_Mercator_coordinate_system
In the file coords.txt you can find the zone (for example 19S).

closed 2017-02-09 by piousgreek

piousgreek, 2017-02-09:
Thank you all, i greatly appreciate the guidance and assistance.  Really cool work, steep learning curve at first.