OpenDroneMap/ODM #473: Global configuration file

dakotabenjamin, 2017-02-08:
There should be a configuration file (YAML) in the ODM directory that has the following:
- default project directory (where all the projects are stored)
- default parameters
- max processors to use
- output types (ply vs las)
- etc.

This would help users who have workflows, among other things.

piousgreek, 2017-02-10:
yes, I was wondering where the default parameters are stored actually.  alternatively, one could place a config file within the project folder or in the images sub directory that would have project specific configuration parameters that would be loaded automatically when ODM is run.

dakotabenjamin, 2017-02-10:
Good idea. I'll keep that in mind.

dakotabenjamin, 2017-03-31:
Merged as of #510

closed 2017-03-31 by dakotabenjamin