OpenDroneMap/ODM #475: IOError: [Errno 2] No such file or directory: './opensfm/reconstruction.json' | opensfm problem

piescie, 2017-02-09:
Hi, 
When I am using my WebODM, I got something like that. I have only 4 cores processor. I've used only 5 photos for processing. What is going wrong?

File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/var/www/data/f740fc86-65b7-42d1-83b6-8d230862cc6a/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "/code/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

dakotabenjamin, 2017-02-09:
Either too few images or too small of an overlap. Try increasing either to get better matches.

dakotabenjamin, 2017-04-03:
Closing, please reopen if you still have issues

closed 2017-04-03 by dakotabenjamin

NeXT1995, 2018-07-02:
@dakotabenjamin  Hello, how much is the data overlap?