OpenDroneMap/ODM #476: Set orthophoto output resolution dynamically using input data

smathermather, 2017-02-09:
One need, rather than output everything in the default 0.05m resolution would be to calculate an appropriate resolution based on the input data. I am not sure the best way to do this, but one way to do this would be to use the info about the height the aircraft is flying (when available), the sensor size and effective focal length, and a low resolution elevation model (or the 3D model and camera positions themselves) to auto calculate a better resolution.

piousgreek, 2017-02-10:
Perhaps it would be beneficial (at least for less experienced users like myself) if ODM could run a quick analysis to recommend various parameters based on the input data or if upon completion of generating all the files it could also output a text file with parameter recommendations should the user elect to run it again.

merkato, 2017-02-21:
Some more configuration possibilities are crucial. If think, target projection in gdal favour ( -t_srs EPSG:xxxx), will be useful. I'm looking for gdal creation options like COMPRESS, TILED too. It'll save my time to manually warp/translate ODM output for use with QGIS or DIVI webgis.

#473 #364

dakotabenjamin, 2017-03-10:
@merkato gdal_translate does not have a t_srs tag. I've added the other creation options, but for now I'm going to leave that one out. Look for a PR in a few minutes, I'd love for you to test.

smathermather, 2017-03-11:
@piousgreek, check out the latest pull request. Dakota has a yaml test configuration file in the pipeline which serves your need here.

pierotofy, 2020-02-23:
This has been implemented a while ago. :+1:

closed 2020-02-23 by pierotofy