OpenDroneMap/ODM #479: question regarding the impact of including overlapped photos from different altitudes

piousgreek, 2017-02-10:
Perhaps this is not the proper venue for these questions, but:

1. How would the accuracy and maximum resolution of the generated orthophotos be affected if ODM were to be applied to a data set that contained images of the same survey flown at two different altitudes (example at 125' and at 400').

2. Can ODM handle mixed data sets of oblique and nadir photos? Do oblique photos need to be tagged or identified in some way with the angle?

3. if a ground control point file is specified, is the gps exif data in the photos ignored?

dakotabenjamin, 2017-02-10:
1. I've not tried it personally but I suppose it would work. @smathermather may know better. 
2. This is ideal for more complex reconstructions- vegetation, buildings, and such. Just make sure you don't have photos with horizon on them. 
3. yes.

smathermather, 2017-02-10:
Great questions all. Mixed altitudes is anticipated to produce better results than single altitude flights, although we haven't specifically tested this.

pulquero, 2017-02-11:
I have had mixed results with mixed data sets and altitudes in the past. I kind of hoped that a high-level photo would help patch together a set of low-level photos, but it didn't seem to work like that. Just seemed to get more confused. This was all pre-the-new-opensfm-stuff, I have much higher hopes with opensfm, but haven't had time to push my big mixed data set through yet.

smathermather, 2017-02-12:
@pulquero -- it's definitely a whole new project with the addition of OpenSfM. If you are willing / able to share test datasets, we'll be happy to help test.

dakotabenjamin, 2017-04-03:
Closing, please reopen if more questions come up :smile:

closed 2017-04-03 by dakotabenjamin