OpenDroneMap/ODM #480: ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

troymyname, 2017-02-10:
Hi,

I am following up from this page: [LINK](https://github.com/OpenDroneMap/OpenDroneMap/issues/334)

I am still getting the error below after following the instructions here: [LINK](https://github.com/troymyname/OpenDroneMap)

from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

Could someone please let me know what I am doing wrong?

troymyname, 2017-02-10:
Please let me know if you need additional details, and I can provide.

troymyname, 2017-02-11:
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/troy/OpenDroneMap-v0_2/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

troymyname, 2017-02-11:
I solved the issue after reading this StackOverflow page: [LINK](https://stackoverflow.com/questions/480764/linux-error-while-loading-shared-libraries-cannot-open-shared-object-file-no-s).

closed 2017-02-11 by troymyname

dakotabenjamin, 2017-02-14:
I'm glad you figured this out. So to understand for future issues, what did you do to solve the problem?

troymyname, 2017-02-15:
I didn't have to do much. I just followed the instructions provided here: [LINK](http://stackoverflow.com/a/21173918/6343136)

That took care of the problem!

Harsha2398, 2017-07-01:
@troymyname Can you please elaborate on what exactly you did please. I'm afraid that link you shared is going over my head as I'm quite new to Ubuntu.

troymyname, 2017-07-04:
Hello and thanks for reaching out. Please follow the instructions in this StackOverflow page: [LINK](https://stackoverflow.com/questions/480764/linux-error-while-loading-shared-libraries-cannot-open-shared-object-file-no-s/21173918#21173918). This should take care of the problem.

sreynolds017, 2018-01-27:
Perhaps the solution is not as intuitive as one might assume.  Obviously, since I'm here and commenting, I've encountered this same error.  I went to the "LINK", and proceeded to run the steps.  The upshot is that I have the LD_LIBRARY_PATH targeted to the location of the libecto.so.0.6 directory, as instructed in the ODM installation instructions.  Nevertheless, the error occurred.  I am not at all sure what to do next.  Perhaps I am missing something obvious?  Suggestions?

steve@steve-Satellite-L775D:~$ cd OpenDroneMap
steve@steve-Satellite-L775D:~/OpenDroneMap$ python run.py --images /home/Steve/ODMProjects/odm_data_aukerman/images /home/Steve/ODMProjects/odm_data_aukerman
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/steve/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory
steve@steve-Satellite-L775D:~/OpenDroneMap$ cd /
steve@steve-Satellite-L775D:/$ sudo find ./ | grep libecto.so.0.6
[sudo] password for steve: 
find: ‘./run/user/1000/gvfs’: Permission denied
./home/steve/OpenDroneMap/SuperBuild/install/lib/libecto.so.0.6.11
./home/steve/OpenDroneMap/SuperBuild/install/lib/libecto.so.0.6
./home/steve/OpenDroneMap/SuperBuild/build/ecto/devel/lib/libecto.so.0.6.11
./home/steve/OpenDroneMap/SuperBuild/build/ecto/devel/lib/libecto.so.0.6
steve@steve-Satellite-L775D:/$ echo $LD_LIBRARY_PATH
:/home/Steve/OpenDroneMap/SuperBuild/install/lib
steve@steve-Satellite-L775D:/$