OpenDroneMap/ODM #482: Linux Debian Jessie 8.0 64bit CMAKE Issue.

TheArceth, 2017-02-12:
After running ./configure.sh

I'm presented with a message

Installing OpenDroneMap Dependencies
Compiling SuperBuild
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.1 or higher is required.  You are running version 3.0.2

-- Configuring incomplete, errors occurred!
Compiling build
CMake Warning at modules/odm_georef/CMakeLists.txt:16 (find_package):
  Could not find a package configuration file provided by "PCL" (requested
  version 1.7) with any of the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/drone/OpenDroneMap/CMakeFiles/CMakeOutput.log".
Configuration Finished

Also during the process I'm seeing this

W: Failed to fetch http://ppa.launchpad.net/george-edison55/cmake-3.x/ubuntu/dists/jessie/main/binary-amd64/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.
Installing Required Requisites

When running:

Traceback (most recent call last):
  File "/home/drone/OpenDroneMap/run.py", line 9, in <module>
    import ecto
ImportError: No module named ecto

Please advise.

dakotabenjamin, 2017-02-14:
That line in [configure.sh](https://github.com/OpenDroneMap/OpenDroneMap/blob/master/configure.sh#L24) should probably have a check for operating system. In your case you need to manually ensure that the CMake version is at least 3.1, then comment that line out and run again. 

The last error you got (no module) is because your environmental parameters are not set correctly

yjmenezes, 2017-02-15:
@TheArceth
for PPA, try this:
Debian Wheezy:
sudo apt-get install python-software-properties

about octo, I have same error:
import ecto

could it be case sensitive MD5 check inside External-Catkin.cmake ?

[ 40%] Performing download step (download, verify and extract) for 'catkin'
-- downloading...
       src='https://github.com/ros/catkin/archive/0.6.16.zip'
       dst='/home/jmenezes/OpenDroneMap/SuperBuild/download/0.6.16.zip'
       timeout='none'
-- [download 69% complete]
[ 85%] Building CXX object libs/tex/CMakeFiles/tex.dir/texture_view.cpp.o
-- [download 70% complete]
-- verifying file...
       file='/home/jmenezes/OpenDroneMap/SuperBuild/download/0.6.16.zip'
-- MD5 hash of
    /home/jmenezes/OpenDroneMap/SuperBuild/download/0.6.16.zip
  does not match expected value
    expected: 'F5D45AE68709CE6E3346FB8C019416F8'
      actual: 'f5d45ae68709ce6e3346fb8c019416f8'
-- Hash mismatch, removing...

I  change from Upper to lower case and  it fix Hash mismatch.
#  URL_MD5           F5D45AE68709CE6E3346FB8C019416F8
  URL_MD5           f5d45ae68709ce6e3346fb8c019416f8

but i  have same error again:
import ecto
ImportError: No module named ecto

yjmenezes, 2017-02-16:
@TheArceth 

try fix three cmakes URL_MD5 to lowercase.

External-Catkin.cmake
External-Ecto.cmake
External-OpenSfM.cmake

[]s
julio

TheArceth, 2017-02-17:
Thank you, I will attempt these on a debian machine but for now reverted to Ubuntu 16.04 LTS.

Before this I did manage to get cmake3.1 installed which was minimum requirement, however could not get it installed with SSL support thus meaning it could not pull items from https git repos. Removing the HTTPS would work however long winded.

dakotabenjamin, 2017-02-21:
The ecto error is due to the environment variable LD_LIBRARY_PATH not being set correctly. Check the instructions in the README to set those properly.

hblanken, 2017-04-24:
For anyone who is interested to run ODM on Debian Jessie, here are all the custom steps I did to get ODM running successfully. Thanks @yjmenezes @dakotabenjamin :

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/linux-debian-jessie-8-0-64bit-cmake-issue/406

closed 2018-03-28 by dakotabenjamin