OpenDroneMap/ODM #484: ARM64: ODM Docker images available

EmbeddedAndroid, 2017-02-15:
If anyone is interested running ODM on ARM64 hardware, I've rebuilt the docker images:

https://hub.docker.com/u/nwdrone/

If your interested in the changes needed to compile ODM under ARM, you can find them here.

https://github.com/EmbeddedAndroid/OpenDroneMap/commits/aarch64

I've been testing these docker images on enterprise grade servers with 64+ ARM64 cores and 128GB of RAM and it has performed very well. 

Anyways, hopefully someone else finds this helpful!

smathermather, 2017-02-16:
That is really exciting, @EmbeddedAndroid. Thank you for that.

EmbeddedAndroid, 2017-02-16:
@smathermather no problem, this is a great project I'd like to see run on ARM.

I've also added node-odm and webodm as aarch64 docker containers.

If your curious, I have a 16 ARM64 core machine with 32GB of memory running node-odm if you want to try a data set on it. I'll leave it running for a few more days.

http://virtio.us:3000

Cheers,

smathermather, 2017-02-19:
Cool. I'll see if I can try it out today or tomorrow.

LucasMM86, 2017-02-19:
I tried your node-ODM, it is great!

smathermather, 2017-02-28:
This is awesome @EmbeddedAndroid. @dakotabenjamin -- nice to see this in the v0.3 milestone.

dakotabenjamin, 2017-02-28:
I want to make its documented somewhere

EmbeddedAndroid, 2017-02-28:
@smathermather @dakotabenjamin I can take a shot at documenting the ARM port, just let me know where you think it should go. 

As for the changes needed to make to get it all to compile, I'll open pull requests for the various projects and link them back to this issue. Once they are all merged, CMake files updated, ODM should just work on ARM.

dakotabenjamin, 2018-03-28:
Picking up on old issues, what's the status of this?

closed 2020-04-14 by smathermather