OpenDroneMap/ODM #486: Core dump by odm_texturing

mbandy, 2017-02-17:
First of all - thanks for this wonderful pierce of software! I have found it very useful.

I encountered an issue on a new build (Ubuntu 14.04). odm_texturing dumped core. As near as I can determine, pmvs failed to load an NVM file. Assuming it's talking about ....

pmvs/recon0/nvmCams.nvm

... then the file exists. However, there is no odm_textured_model.obj file in odm_texturing, and this may be the problem. 

Any help would be much appreciated. Relevant output is below, and the full output is attached.

(https://github.com/OpenDroneMap/OpenDroneMap/files/782364/odm_failure_output.txt)
 
Thank you!

`[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /home/inti/drone_projects/la_72303/odm_texturing/odm_textured_model.obj
[DEBUG]   Generating .nvm file from pmvs output: /home/inti/drone_projects/la_72303/pmvs/recon0/nvmCams.nvm
[DEBUG]   running /home/inti/OpenDroneMap/SuperBuild/install/bin/texrecon /home/inti/drone_projects/la_72303/pmvs/recon0/nvmCams.nvm /home/inti/drone_projects/la_72303/odm_meshing/odm_mesh.ply /home/inti/drone_projects/la_72303/odm_texturing/odm_textured_model -d gmi -o none
/home/inti/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Feb 16 2017, 18:55:07)
Load and prepare mesh:
PLY Loader: comment PCL generated
Reading PLY: 100015 verts... 199862 faces... done.
Generating texture views:
NVM: Loading file...
NVM: Number of views: 109
NVM: Number of features: 0
terminate called after throwing an instance of 'util::FileException'
  what():  Permission denied
Aborted (core dumped)
Traceback (most recent call last):
  File "/home/inti/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/inti/OpenDroneMap/scripts/mvstex.py", line 106, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/inti/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134`

mbandy, 2017-02-17:
See #437  - using --use-opensfm-pointcloud resolved this issue.

closed 2017-02-17 by mbandy

smathermather, 2017-02-17:
Excellent! We do need to switch this to the default setting soon.