OpenDroneMap/ODM #487: Segmentation Fault (Core Dumped)

TheArceth, 2017-02-17:
Running on 12 Core Xeon with 32GB memory,
Ubuntu 16.04 LTS

Install went fine no problems. Running with default settings on 500 images Geo Tagged JPG's

---- Initial: 0 secs ----
Total pass fail0 fail1 refinepatch: 1953174 115859 1415117 422198 538057
Total pass fail0 fail1 refinepatch: 100 5.93183 72.4522 21.616 27.5478
Expanding patches...Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/home/drone/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/drone/OpenDroneMap/scripts/pmvs.py", line 73, in process
    (context.pmvs2_path, tree.pmvs_rec_path))
  File "/home/drone/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

TheArceth, 2017-02-17:
Furthermore:

This is running within a XenServer Virtual Image with dynamic memory allocation. Set this to fixed memory allocation and re-running with clean project, to see if this fixes the issue.

j6k4m8, 2017-02-19:
Experiencing the same issue on Ubuntu 16.04, 12GB mem on a relatively small dataset (so i don't think it's an out-of-memory issue).

dakotabenjamin, 2017-02-21:
Also try `--use-opensfm-pointcloud` to use our newer algorithm

TheArceth, 2017-03-24:
Good Afternoon,

I changed this from Dynamic to a set memory allocation, dropped the memory and cores in use to a smaller amount and it ran fine.

So initially I thought this a problem with the system however it looked more like an issue with the images. We tried running it again with the switch you provided however we began to see some very strange darkness turn up in parts of the photo.

closed 2017-03-24 by TheArceth