OpenDroneMap/ODM #488: ERROR RUNNING ODM WITH DJI FC330 images

Davidlapalma, 2017-02-17:
Hello, Have someone the same error?, and how can I fix the fail?
the code is the next:
davidgc@davidgc-MacBookPro:~$ cd /home/davidgc/Escritorio/OpenDroneMap 
davidgc@davidgc-MacBookPro:~/Escritorio/OpenDroneMap$ python run.py --project-path /home/davidgc/Escritorio/OpenDroneMap/project 
[INFO]    Initializing OpenDroneMap app - Fri Feb 17 13:08:06  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /home/davidgc/Escritorio/OpenDroneMap/project/images
Traceback (most recent call last):
  File "run.py", line 50, in <module>
    plasm.execute(niter=1)
  File "/home/davidgc/Escritorio/OpenDroneMap/scripts/dataset.py", line 63, in process
    path_files
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 251, in map
    return self.map_async(func, iterable, chunksize).get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 567, in get
    raise self._value
IOError: /home/davidgc/Escritorio/OpenDroneMap/project/images/._Captura de pantalla 2017-02-16 a las 21.28.11.png: The file contains data of an unknown image type

TheArceth, 2017-02-17:
Not an expert, but I would assume the following

JPG and not PNG
The period before the _Captura would seem to note a file type of some sort?

Davidlapalma, 2017-02-17:
I have resolve that error but now it appear the next:
Traceback (most recent call last):
  File "/home/davidgc/Escritorio/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/davidgc/Escritorio/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/home/davidgc/Escritorio/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/home/davidgc/Escritorio/OpenDroneMap/project/opensfm/reconstruction.json'
[ERROR]   quitting cause: 
	PYTHONPATH=/home/davidgc/Escritorio/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/davidgc/Escritorio/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/davidgc/Escritorio/OpenDroneMap/project/opensfm
returned with code 256.

An error occurred. Check stdout above or the logs.

Davidlapalma, 2017-02-17:
I think that the error happen because the images are taken to different heights

dakotabenjamin, 2017-02-21:
OpenSfM can fail to reconstruct for various reasons. What % overlap did you fly for your images?

Davidlapalma, 2017-02-21:
100%

Davidlapalma, 2017-02-21:
I think that this is the problem that there are images that are 100% of others

dakotabenjamin, 2017-02-21:
Maybe. Can you share your images? You can either post the link here or privately: dmb2[at]clevelandmetroparks[dot]com

Davidlapalma, 2017-02-21:
yes sure

Davidlapalma, 2017-02-21:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Davidlapalma, 2017-02-21:
[image]

Davidlapalma, 2017-02-21:
i'm having problems so I'n going to send via email to you

closed 2017-02-21 by Davidlapalma

dakotabenjamin, 2017-02-21:
OK

dakotabenjamin, 2017-02-21:
Here's what I recommend:
1. Fly again in a criss-crossing pattern to get different angles for reconstruction to find matches
2. Use [this calibration tool](https://github.com/dakotabenjamin/CameraCalibration) to remove distortion from the images. Note that you can run on Windows but the exif data will not be copied over (have to do so manually)

Davidlapalma, 2017-02-21:
ok thanks