OpenDroneMap/ODM #489: Is possible implement the program on GPU with CUDA?

Davidlapalma, 2017-02-17:
I have really insterested in take the code and implemente on a GPU NVIDIA to run the program faster.
It is possible, can you any idea about that?

smathermather, 2017-02-17:
This is in the wish list -- just waiting on time and funding to implement. Cheers!

Davidlapalma, 2017-02-17:
I have the intention to do a final project on my university, maybe its no much work I can try it

Davidlapalma, 2017-02-17:
I will try it in a Nvidia Jetson

smathermather, 2017-02-18:
👍

gateway, 2017-02-27:
An issue is opened up here https://github.com/OpenDroneMap/OpenDroneMap/issues/92 and since Open Drone Map uses OpenSfM they have a old pull request that could use some help https://github.com/mapillary/OpenSfM/pull/49  OpenCV has supported gpu's for some time now.

dakotabenjamin, 2017-02-28:
Thanks for digging in @gateway I know that contributor is currently working on their PhD so there's a good reason that pull request remains incomplete. Perhaps this is a good starting place for @Davidlapalma

Davidlapalma, 2017-03-13:
thank you @gateway, I will start trying that

PeterSprague, 2017-11-08:
Any progress on this?

I have a working nvidia-docker ODM-cuda-8.0-opencv-3.3.1 image to use as a base now.  What are the next steps to push this forward?

OpenSfM is not utilizing the GPU yet though OpenDroneMap/OpenCV is built against Cuda:
$ nvidia-smi
Tue Nov 7 09:50:17 2017
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.90 Driver Version: 384.90 |
|-------------------------------+----------------------+----------------------+
| GPU Name Persistence-M| Bus-Id Disp.A | Volatile Uncorr. ECC |
| Fan Temp Perf Pwr:Usage/Cap| Memory-Usage | GPU-Util Compute M. |
|===============================+======================+======================|
| 0 GeForce GTX 1080 Off | 00000000:05:00.0 On | N/A |
| 27% 34C P8 8W / 180W | 451MiB / 8112MiB | 4% Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes: GPU Memory |
| GPU PID Type Process name Usage |
|=============================================================================|
| 0 1784 G /usr/libexec/Xorg 414MiB |
| 0 2415 G /usr/lib64/firefox/firefox 3MiB |
| 0 20753 G /proc/self/exe 29MiB |
+-----------------------------------------------------------------------------+

I thought OpenSfM would pick up the GPU if OpenCV was built against Cuda? 

mapillary/OpenSfM#49 mentions "First working version for ORB gpu. Some numbers: HAHOG takes 1sec aprox. in Lund images, ORB-GPU 0.008s aprox."  Is it even feasible to move to ORB-GPU within the ODM stack?  Is this the best approach from ODM perspective?

PeterSprague, 2017-11-08:
Any progress on this?

I now have a working nvidia-docker ODM-cuda-8.0-opencv-3.3.1 image to use as a base.  What are the next steps to push this forward?

OpenSfM is not utilizing the GPU yet though OpenDroneMap/OpenCV is built against Cuda:
$ nvidia-smi
Tue Nov 7 09:50:17 2017
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.90 Driver Version: 384.90 |
|-------------------------------+----------------------+----------------------+
| GPU Name Persistence-M| Bus-Id Disp.A | Volatile Uncorr. ECC |
| Fan Temp Perf Pwr:Usage/Cap| Memory-Usage | GPU-Util Compute M. |
|===============================+======================+======================|
| 0 GeForce GTX 1080 Off | 00000000:05:00.0 On | N/A |
| 27% 34C P8 8W / 180W | 451MiB / 8112MiB | 4% Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes: GPU Memory |
| GPU PID Type Process name Usage |
|=============================================================================|
| 0 1784 G /usr/libexec/Xorg 414MiB |
| 0 2415 G /usr/lib64/firefox/firefox 3MiB |
| 0 20753 G /proc/self/exe 29MiB |
+-----------------------------------------------------------------------------+

I thought OpenSfM would pick up the GPU if OpenCV was built against Cuda? 

mapillary/OpenSfM#49 mentions "First working version for ORB gpu. Some numbers: HAHOG takes 1sec aprox. in Lund images, ORB-GPU 0.008s aprox."  Is it even feasible to move to ORB-GPU within the ODM stack?  Is this the best approach from ODM perspective?

PeterSprague, 2017-11-08:
Any progress on this?

I now have a working nvidia-docker ODM-cuda-8.0-opencv-3.3.1 image to use as a base [https://github.com/OpenDroneMap/OpenDroneMap/issues/694].  What are the next steps to push this forward?

OpenSfM is not utilizing the GPU yet though OpenDroneMap/OpenCV is built against Cuda:
$ nvidia-smi
Tue Nov 7 09:50:17 2017
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.90 Driver Version: 384.90 |
|-------------------------------+----------------------+----------------------+
| GPU Name Persistence-M| Bus-Id Disp.A | Volatile Uncorr. ECC |
| Fan Temp Perf Pwr:Usage/Cap| Memory-Usage | GPU-Util Compute M. |
|===============================+======================+======================|
| 0 GeForce GTX 1080 Off | 00000000:05:00.0 On | N/A |
| 27% 34C P8 8W / 180W | 451MiB / 8112MiB | 4% Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes: GPU Memory |
| GPU PID Type Process name Usage |
|=============================================================================|
| 0 1784 G /usr/libexec/Xorg 414MiB |
| 0 2415 G /usr/lib64/firefox/firefox 3MiB |
| 0 20753 G /proc/self/exe 29MiB |
+-----------------------------------------------------------------------------+

I thought OpenSfM would pick up the GPU if OpenCV was built against Cuda? 

mapillary/OpenSfM#49 mentions "First working version for ORB gpu. Some numbers: HAHOG takes 1sec aprox. in Lund images, ORB-GPU 0.008s aprox."  Is it even feasible to move to ORB-GPU within the ODM stack?  Is this the best approach from ODM perspective?

dakotabenjamin, 2017-11-08:
The path to getting it in odm:
1. fix conflicts in the OpenSfM 
2. Get it merged by mapillary
3. Make a PR in ODM to pull in latest opensfm changes

If there was trouble getting it pulled into opensfm, then there are ways to patch it into odm otherwise.

I think edgarriba has moved on from this project, probably got busy with their PhD (understandably).

PeterSprague, 2017-11-08:
Is this the best approach, aka ORB-GPU, to use from the ODM perspective?  Or is a fresh approach the better choice?  Most recent work is from 2015.

beedawg1985, 2019-06-20:
Is CUDA supported now or likely to be supported in the near future?

smathermather, 2019-06-24:
CUDA support was dependent upon the expiration of the SIFT patent (at least for it to be an easily doable/fundable item). It is now possible, but requires either someone to take it on as a project or find funding.

seangrant82, 2020-02-10:
Just wanted to bump this to see if there is anyone out there that can take this on (I would if I could) or has tried to since last post. I think this would be a HUGE win to get GPU support.

smathermather, 2020-02-11:
Nothing yet, but getting SIFT into the tool chain will help, and the patent expires soon: see https://github.com/OpenDroneMap/ODM/pull/1069

smathermather, 2020-04-14:
Duplicate. Closing.

closed 2020-04-14 by smathermather