OpenDroneMap/ODM #491: ODM not matching all images

luizzod, 2017-02-21:
I'm trying to make an orthophoto out of 720 images taken by a drone with the GPS information in the exif. 

I run the following command:
`python run.py --project-path ~/FincaMM/ --resize-to 4000 --use-opensfm-pointcloud`

This is what comes out (the image has been resized to 10% of original image):
[image]
I get the same results if ran with pmvs instead of opensfm.

By comparison, this is what Agisoft renders with the same set of images (also 10% of original):
[image]

Any reason why they differ so much with this set? For another set using less images and not taken using waypoints I didn't have this problem.

dakotabenjamin, 2017-02-21:
Interesting... I assume the point cloud is also that size? I would advise increasing `--min-num-features` and resizing to 2000.

dakotabenjamin, 2017-02-21:
Sometimes when the image is high resolution, features become difficult to match. It is a paradoxical problem.

smathermather, 2017-02-21:
Alternatively, try the resize as default 2400.

dakotabenjamin, 2017-02-27:
@luizzod did this work?

luizzod, 2017-02-27:
I did try to run it at 2000 num features with resize 2000 however when it was doing depthmap cleaning it would stay there like if it hanged. I tried this several times and it seems to hang at that point. Im increasing my ram to 32gb and switching to a better processor. Will update once I run it again.

luizzod, 2017-02-27:
I meant that I increased the --min_num_features by 2000. Actually ran it at 8000 first and then 6000 but I ended up with the same hang. It simply stops processing any more images and when I look up the cpu usage it is not being used but memory is at full capacity. 
My specs are i5-4440 with 16Gb Ram, running on Gnome Ubuntu 16.04 LTS. Im doubling the ram and switching the processor to an i7-4790k later this week.

dakotabenjamin, 2017-02-28:
When it hangs like that you need to reduce the number of cores being used in software, using `--opensfm-processes n`, with n= less than the maximum number of cores. I usually do about 3/4 the total. This is a known bug being worked on.

luizzod, 2017-03-16:
So I managed to get my computer back online and I've run a dataset of around 300 images both with pmvs and opensfm. This is a new dataset of the same field. 

Running pmvs using 8000 number of features:
[image]

Running again but with opensfm:
[image]
 

These are better results than before however they are still incomplete. 
I also used Agi for this dataset, below is what came out.
[image]

smathermather, 2017-03-16:
Is this a dataset that can shared? We're looking for problematic test datasets.

luizzod, 2017-03-16:
Yes. Currently uploading the dataset. Will post when link is ready.

luizzod, 2017-03-16:
The dataset can be found in the following link:
https://1drv.ms/f/s!AlHF70N2QL1_g60fwi5K8rXW2zV5hQ

dakotabenjamin, 2017-03-23:
I'm running the dataset now, will let you know how it goes!

dakotabenjamin, 2017-03-23:
Using a QGIS plugin called Photo2Shape I looked at the camera locations for your linked dataset. It looks like there's uneven overlap which I'm guessing is the reason for the incompleteness. Are you running manual flights? 

Also, there's something strangely gridlike about the gps locations, how are you georeferencing the images?

[image]

dakotabenjamin, 2017-03-23:
For reference, here's a dataset from a flight flown with autopilot on a grid:

[image]

dakotabenjamin, 2017-03-23:
My result, default params: 

[image]

luizzod, 2017-03-23:
The images were taken using point to point autopilot with the autel x-star. The incompleteness is because at those points I had to switch the battery and when I set the autopilot again it doesn't continue from last point rather I manually have to set the point. There's two different flight paths on the left and one that starts on the right of that grid on the sattelite image. It seems ODM on default is only taking the first path into account. Opensfm only uses flight path 2 and flight path 3 only shows up in agi. 
The images are georeferenced automatically with the GPS information from the drone.

dakotabenjamin, 2017-03-23:
It's not that OpenSfM is only doing flight path 1, the software is not so deterministic. It's able to create partial reconstructions of the whole ROI:
```
2017-03-23 17:09:16,366 Reconstruction 1: 66 images, 14518 points
2017-03-23 17:09:16,366 Reconstruction 2: 34 images, 6231 points
2017-03-23 17:09:16,366 Reconstruction 3: 20 images, 5894 points
2017-03-23 17:09:16,366 Reconstruction 4: 17 images, 3984 points
2017-03-23 17:09:16,366 Reconstruction 5: 13 images, 3615 points
2017-03-23 17:09:16,367 Reconstruction 6: 12 images, 1796 points
2017-03-23 17:09:16,367 Reconstruction 7: 10 images, 634 points
2017-03-23 17:09:16,367 Reconstruction 8: 4 images, 262 points
2017-03-23 17:09:16,367 Reconstruction 9: 2 images, 89 points
2017-03-23 17:09:16,367 Reconstruction 10: 2 images, 63 points
2017-03-23 17:09:16,367 Reconstruction 11: 2 images, 60 points
2017-03-23 17:09:16,367 Reconstruction 12: 2 images, 52 points
2017-03-23 17:09:16,367 Reconstruction 13: 2 images, 52 points
2017-03-23 17:09:16,367 14 partial reconstructions in total.
```

But only the first reconstruction is used for the model. This is because there are too few matches between tracks. ODM is somewhat conservative in this regard.

dakotabenjamin, 2017-04-03:
I'm going to close this for now, feel free to reopen if you have lingering questions.

closed 2017-04-03 by dakotabenjamin