OpenDroneMap/ODM #492: Tone Mapping parameter missing for mvs_texturing

pierotofy, 2017-02-21:
While digging through mvs_texturing's code, I noticed that we currently don't have a way to enable the tone mapping feature, here:

https://github.com/nmoehrle/mvs-texturing/blob/master/apps/texrecon/arguments.cpp#L69

Which I think is directly related to the "bland" colors you sometimes observe as output on the orthophotos: 
[image]

Notice the white areas near trees.

This seems directly related to the tone mapping feature, which is disabled by default: https://github.com/nmoehrle/mvs-texturing/issues/24

dakotabenjamin, 2017-02-21:
I will see if exposing this parameter will help.

dakotabenjamin, 2017-02-22:
https://github.com/dakotabenjamin/OpenDroneMap/tree/tone-mapping-param 

Note that this doesn't include an update to MVS-Texturing. You will have to pull #471 changes in or update manually. 

I don't have a dataset on hand to find any tangible differences between 'gamma' and 'none' so I'm hoping you can test that. Also, I was unsure whether to use the mvs-tex args (gamma and none) or just a boolean flag, but can easily switch if you think one is preferable over the other.

pierotofy, 2017-02-25:
Thank you @dakotabenjamin! I will do some testing and report my results.

pierotofy, 2017-02-25:
Mm, no significant improvements over the faded trees. This is what I got out with the latest mvs texturing and tone mapping set to gamma:

[image]

I think the tone mapping option should be included, but perhaps have it set by default to none. Leaving it as a string is fine by me.

pierotofy, 2017-02-25:
Settings tone mapping to `none` with the latest mvs texturing resulted in much nicer results:

[image]

So while just upgrading to the latest mvs texturing code will fix the problem, it might be beneficial for some users to expose the tone mapping option. But better to leave the default to `none`.

pierotofy, 2017-03-06:
I'll close this since it has been taken care of as part of #499

closed 2017-03-06 by pierotofy