OpenDroneMap/ODM #497: Stitching large image dataset of around 90GB

shahvicky, 2017-02-26:
Hi Team,

We have a client who has more than 1000 acres of palm plantation. They take images from the drone and the total size of images generated is approx 90GB. The number of images is more than 10000.
The images are taken from a height of around 250ft. 
How to go ahead with such a large dataset to complete the stitching in the minimum time. 
I mean, how to divide the dataset and then process. Or if there is any other way. Please help.

yjmenezes, 2017-02-26:
@shahvicky 

i think you could divide your area into rectangles blocks using some spatial data bank. Spatialite or MariaDB or PostGIS.

1- Use SQL query ST_Within to select the photos inside a block + an extra overlap  buffer area .

https://mariadb.com/kb/en/mariadb/st-within/
http://www.gaia-gis.it/gaia-sins/spatialite-sql-4.3.0.html

2- use GDAL to clip the processed image blocks  and to merge.

julio

piousgreek, 2017-03-01:
@shahvicky 

This may give you some additional guidance:
https://www.youtube.com/watch?v=vmGC_W9besQ

shahvicky, 2017-03-02:
@yjmenezes @piousgreek
Thanks for the prompt response. 
But the images that I have does not contain geolocation data. They are raw .cr2 images. Can you help to divide into small dataset without GPS data?

yjmenezes, 2017-03-02:
@piousgreek 

1- do you have some systematic data behavior, like photo names, flight strip number + photoname,  anything we could use to organize the subsets ?
2- do you have any gps record we could correlate  to photos somehow ?

please, take a look:
https://github.com/LibRaw/LibRaw/tree/0.18-stable

[]s
julio

shahvicky, 2017-03-02:
@yjmenezes 

1) We have just images with names in serial order as taken by the drone. 
eg: img01078.cr2, img01079.cr2, img01080.cr2 etc. 
I tried to see the EXIF at Jeffrey's Image Metadata Viewer(http://exif.regex.info/exif.cgi)
The below is the result
[Jeffrey Friedl's Image Metadata Viewer.pdf](https://github.com/OpenDroneMap/OpenDroneMap/files/814869/Jeffrey.Friedl.s.Image.Metadata.Viewer.pdf)
please check if we can make out any useful information out of it.
In the meanwhile, I will also check if we can get the gps record to relate to photos. 

2) I also tried stitching a dataset of around 170 images, all without any geolocation data but the process failed with error message that the images require geolocation data. Is it possible to just stitch the images without gps data? I do not need geotiff, but just a stitched image to run some ndvi calculation on it. 

Thanks,
Vivek

dakotabenjamin, 2017-03-02:
@shahvicky we should have better large dataset support by the summer. 

In the meantime,  if your drone records/tracks GPS then it may be possible to tag the photos using the timing of the GPS track and the photos. I have done this before, I'll dig up the code if I can find it. 

for 2. What error message are you getting? Without geodata, there is no way for the software to know which way is up in order to frame the resulting image from the point cloud.

piousgreek, 2017-03-03:
i have had some success geotagging images with Geosetter as well as Mission Planner.

shahvicky, 2017-03-03:
@dakotabenjamin 

Can you please provide the code for tagging the photos with the GPS. I can try and share the result as soon as I get the file with timing+GPS data. 

For 2) I was getting the following error while processing images without the geotag. I am using WebODM for the task. 

`[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /var/www/data/dd71c16d-07f4-45f7-b1b9-7484d9a39f2a/odm_georeferencing/coords.txt
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file. Error: Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.

[WARNING] Georeferencing failed. Make sure your photos have geotags in the EXIF or you have provided a GCP file. 
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running ODM Orthophoto Cell
[DEBUG]   running /code/build/bin/odm_orthophoto -inputFile /var/www/data/dd71c16d-07f4-45f7-b1b9-7484d9a39f2a/odm_texturing/odm_textured_model.obj -logFile /var/www/data/dd71c16d-07f4-45f7-b1b9-7484d9a39f2a/odm_orthophoto/odm_orthophoto_log.txt -outputFile /var/www/data/dd71c16d-07f4-45f7-b1b9-7484d9a39f2a/odm_orthophoto/odm_orthophoto.png -resolution 20.0  -outputCornerFile /var/www/data/dd71c16d-07f4-45f7-b1b9-7484d9a39f2a/odm_orthophoto/odm_orthophoto_corners.txt`

For now, is there any limit on the number of images that can be given for processing. I mean what is the maximum number of images recommended not to exceed while processing.

LucasMM86, 2017-03-03:
@shahvicky  did you try to open your textured mesh with Meshlab? I mean the texturing process is complete, so you have already a textured mesh.

shahvicky, 2017-03-03:
@LucasMM86 

I am using WebODM. From where can I get the textured mesh? Can you help me with this. ?

yjmenezes, 2017-03-03:
@shahvicky 

Do you have the flight plan or some drone flight logs ?

I think it is possible to write the Exiftags using exiftools.
I have done this using a GNU/Linux script,  to process scanned analog photos.
A command line like this:

exiftool -GPSLatitude=ddd.ddddd -GPSLatitudeRef=S -GPSLongitude=ddd.dddd
-GPSLongitudeRef=W -GPSAltitude=mmm.mm your_image.jpg

As the photos have sequential number we can try to solve.
Please, would be possible to send me this informations:

we can try to build a small block, for this we will need:
a) the mean flight altitude.
b) measure the  img01078.cr2 image center (long/lat)  from google
c) measure the img01079.cr2  image center (long/lat)  from google
d) measure the img10xx.cr2 at the end of the flight strip;

i will try, assuming all are over the same strip and taken at constant speed, compute all others image centers.
This way ODM can align the photos, i guess.

[]s
julio
PS: would be possible to send me one image pair, img01078.cr2, img01079.cr2

pierotofy, 2017-03-03:
@shahvicky once the task is completed in WebODM, press the "Download Assets" --> "All" button. One of the directories contains the textured mesh.

dakotabenjamin, 2017-03-03:
@shahvicky https://github.com/mapillary/mapillary_tools/blob/master/python/geotag_from_gpx.py is what I used. It will probably need to be modified to your specific needs though.

shahvicky, 2017-03-03:
Thanks for the support. 
@pierotofy 

Since the task failed, I am not able to see the "Download Assets" button. Only "Edit", "Restart" and "Delete" buttons are visible for that task.
The same button is available for the completed tasks.

@dakotabenjamin 

What is the maximum number of images/dataset recommended not to exceed while processing?

Is there a development github branch to tryout the large dataset feature?

dakotabenjamin, 2017-03-03:
There is no hard limit. It depends on cpu and ram as well as image size and image number. I usually will not try to process more than 700 images on my server (16vCPU and 44 GB ram)

The large dataset feature is in its infancy, no code exists yet.

smathermather, 2017-03-03:
Also a thing to note: if you have no geo tagging on your imagery, you will be able to process fewer images.

Expect draft large dataset feature by May.

dakotabenjamin, 2017-03-03:
I think what @smathermather means is that the matching process uses k-nearest neighbors to find nearby cameras to match with. Otherwise it does pairwise matching. Big O is something like 20n vs n^2 (iirc)

jo-chemla, 2017-03-03:
Following PMVS, Y. Furukawa published a paper for CMVS in 2010, aimed at that specific purpose of splitting a large dataset processing without needing to worry about memory limit. P. Moulon (OpenMVG) then refreshed the code base here : https://github.com/pmoulon/CMVS-PMVS Maybe it can serve as a first start

smathermather, 2017-03-03:
CMVS-PMVS was a starting point for the project! Short term, we did trade scalability for density of output by switching away from it in favor of OpenSfM's dense matching solution.

jo-chemla, 2017-03-03:
Thanks for that clarification. I think there are solutions to build a denser cloud from a PMS/CMVS output, will need to dig into that ( https://github.com/simonfuhrmann/mve/issues/221 )

dakotabenjamin, 2017-04-03:
All, Please check this blog post for more info regarding future scaling improvements: https://smathermather.com/2017/03/08/scaling-opendronemap-necessary-and-fun-next-steps/

dakotabenjamin, 2018-03-28:
Split merge has been implemented! Check the docs for more info on how to use: docs.opendronemap.org

closed 2018-03-28 by dakotabenjamin

navjot5263, 2020-04-19:
@shahvicky 
hello sir! 
Could you please tell me how you did stitching by using gps data . I am also doing the same but unable to figure out the method to stitch them.I am having data of drone images of agriculture.Can you share your code with me.

shawnngtq, 2020-08-21:
@shahvicky,

In the end did you managed to do that on your dataset? I was wondering if you can share more about:

1. Hardware specs
2. Time taken

Would be perfect if you have a blog / doc / note to share!