OpenDroneMap/ODM #500: Align of resulting mosaic (in case of multispectral imaging)

ivopavlik, 2017-02-28:
Each band of multispectral sensors (e.g. Micasense Rededge, Sequoia) is taken with separate lens, where different angle for each band is appied (each band image is in separate file). When sources images are merged to multiband image, there bands are significantly shifted each other. Therefore, each band must be processed separately in case of mosaicking. Unfortunately, when merging resulting mosaics, similar shift is present:
[image]
Possibility of aligning resulting mosaics would be appreciated.

Thanks.

smathermather, 2017-02-28:
This is related to a pending feature request from @newcomb-d

LucasMM86, 2017-11-22:
Hello, i share my experience with datasets from different cameras. The cameras are a survey 2 RGB and a Survey 2 NIR. The approach is to run the opensfm and mesh processes with both datasets together. Then run mvs_texturing separately for each dataset.
The attached file shows the steps for doing this.
[Steps.pdf](https://github.com/OpenDroneMap/OpenDroneMap/files/1493585/Steps.pdf)

BORDENEUVE, 2017-11-24:
Hello everyone I wonder if it would not be possible to introduce in the georeferencing photos a correction by taking into account the offset of the optics.for the treatment of photos for me the best solution is to have a mosaic of photos per tape and perform the treatment (rgb, ndvi, ndvre) with qgis

BORDENEUVE, 2017-11-25:
Hello
could anyone test this manipulation
assemble the 5 layers of the same images (mica sense red edge) with qgis ,then make a mosaic of several images of the same type with ODM

smathermather, 2018-04-23:
Very interesting, @bcjarrett -- any code you can share? Maybe that could be a contributors script to prep the data before feeding through ODM.

bcjarrett, 2018-04-24:
@smathermather Unfortunately, the process is pretty destructive and strips any existing EXIF data, making further processing challenging. I'll update here if I ever end up with a more fully realized solution.

bcjarrett, 2018-04-24:
That makes complete sense, and is what I do for orthomosaic projects. With accurate GCPs everything ends up very well positioned. But, if for example, there is tree cover that causes blurs in the mosaic and I need to look at individual images so I can see the forest floor clearly I need to be able to line the images up to generate an NDVI.

LucasMM86, 2018-04-24:
I am not sure if we are speaking of the same thing. The process i am speaking about doesnt need GCPs. If you are trying to align individual images Ok this is another thing.

bcjarrett, 2018-04-24:
I am speaking of individual images

smathermather, 2018-04-25:
@LucasMM86 -- this makes perfect sense. Now we just need to script this up... .

LucasMM86, 2018-04-25:
Hello, i did small python scripts to do the files renaming and to change the "reconstruction.nvm" file, but i didnt make a script for the whole process. I will try to find them. But anyway i am not sure if they are "well" written, my programming skills are limited... Regards, Lucas

smathermather, 2018-04-25:
Poorly written is fine, @LucasMM86. We can always polish them up as needed. Cheers!

BORDENEUVE, 2018-04-25:
Hello everyone
to assemble a photo mosaic I found this software MIC MAC, I can not test it yet
  first case make him assemble an image with several bands and then cross the images with QGIS
second case with this software to assemble mono-band photo mosaics that we would overlay with QGIS

MIC MAC software link https://micmac.ensg.eu/index.php/Home
MIC MAC forum discussion multispectral image http://forum-micmac.forumprod.com/photogrammetry-with-sequoia-multispectral-imaging-t1510.html

smathermather, 2018-04-26:
@BORDENEUVE -- MIC MAC is an awesome project, and I love how focused they are on research questions. Definitely a lot to learn from their work. Also, AFAIK, their license is compatible with ours, so mixing is a possibility... .

LucasMM86, 2018-09-21:
Hello,
here i found the scripts i mention in my post on this thread. First I use this python script in order to rename each set of images (all the RGB images are called RGB01.JPG, RGB02.JPG, ... and all the NIR images are called NIR01.JPG, NIR02.JPG, ...). 
"
import glob, os

path='./imagesRGB'

NIR_or_RGB='_RGB'

for file_ in os.listdir(path):
    print file_
    os.rename(path+'/'+file_,path+'/'+file_+NIR_or_RGB+'.JPG')
"

Then put all NIR and RGB images in the same folder called "images". Run ODM.
Now we take from the opensfm/reconstruction.NVM file only the RGB images. I use this script for doing this:
"
import os

band='RGB'  # RGB or NIR
path='./opensfm'

file_=open(path+'/reconstruction.nvm','r')
file_mod=open('./reconstruction_'+band+'.nvm','w')
b=1
cuenta_fotos=0
banda_in_a=0
while b==1:
    a=file_.readline()
    if band in a:
        banda_in_a=1
        cuenta_fotos=cuenta_fotos+1
    else:
        banda_in_a=0
    if 'undistorted' in a:
        linea_no_foto=0
    else:
        linea_no_foto=1
    if banda_in_a or linea_no_foto:
        file_mod.write(a)
    #b=b+1
    if a=='':
        b=0
    print a

file_.close()
file_mod.close()
print 'number of images: '+str(cuenta_fotos)
"
Now, runinig ODM again, it happens that the texturing and the orthophoto steps use only the RGB images. In that way we obtain the RGB orthomosaic.
Now we do the same process (using the last script) to generate a reconstruction.nvm file only with NIR images and run ODM again, and obtain the NIR orthomosaic, which is perfect aligned with the RGB ortho...

Note: this works only with opensfm for sparse point cloud and not pmvs, since pmvs changes the names of the images in reconstruction.nvm. (I think opensfm is the default now).

I belive this could be better scripted, with a single script runing "on top" of ODM.

smathermather, 2018-09-21:
It seems this has a very similar approach to split merge and could be integrated when that process gets rewritten. Thoughts @pierotofy @dakotabenjamin?

pierotofy, 2018-09-22:
Somewhat similar, involves processing two datasets separately (from the beginning, split-merge does the split after opensfm's reconstruction) and then merging the results.

smathermather, 2018-09-22:
Correct. The underlying problem is essentially the same, the current approach for addressing it does differ in some details.

smathermather, 2018-09-23:
What I think is particularly exciting and compelling about these related solutions is that we need to make split-merge more core to the pipeline _and_ we have this broader set of problems to solve in multi-spectral that overlaps the split-merge problem in substance and likely in implementation.

I wonder too if this points at the solution for >8-bit datasets -- we probably don't need to do matching in 16-bits, but we do need to use the output from an SfM solution to help us provide appropriate reconstructions.

Unless I'm missing something, this means a full 16-bit solution simply (ha!) requires a 16-bit texturing solution. For those interested in quantitative outputs from these data, e.g. when flying a farm field or similar, a lumitexel approach might be necessary.

spifftek70, 2018-10-09:
As normal jpg files support only 3 bands, have you considered using a simple band2band registration script to merge the four individual band images into one jpeg2000 file and then stitching those files?

smathermather, 2018-10-09:
The issue is that each band doesn’t really align, as the cameras are offset physically from each other. So it’s less a question of how many bands, and more a question of the alignment of the bands.

spifftek70, 2018-10-10:
But if the bands are properly aligned, then ODM will stitch jpg2000?

smathermather, 2018-10-10:
I suspect not. I don't think OpenSfM, the underlying library, is able to use jpg2000, but I haven't investigated or tested myself.

spifftek70, 2018-10-10:
I know the initial execution in docker doesn't recognize jp2 or jpk extensions.

BORDENEUVE, 2018-10-10:
Hello
by measuring the positioning between the different objectives according to the axes (front back and left right) could not your pitch corrected the shift between the different bands

smathermather, 2018-10-10:
Correct, and I believe this is how it’s handled in agisoft.

BORDENEUVE, 2018-10-11:
for agisoft I do not know but it exists for Pix4D

spifftek70, 2018-10-11:
If you had such a code for band2band registration, you still have the underlining issue of OpenSfM.  Is jpg the only format it takes?

smathermather, 2018-10-11:
It takes PNG as well, but AFAIK, only 8-bit PNG. From a structure from motion perspective, this isn't an major issue -- we can do all the matching and SfM steps in 8-bit under most conditions without any real loss of quality, but the subsequent texturing steps as 8-bit are a problem that needs solved.

kikislater, 2018-10-11:
Watch Pix4fields, it's not SFM. It's basic old photogrammetric process with stitching. Just watch what they use  : 
[image]

[image]
[image]

[image]

Pretty sure it uses gdal + python to put image on the ground (we have omega phi kappa from exif) then stitch them with opencv => https://docs.opencv.org/3.4.2/d8/d19/tutorial_stitcher.html

Watch @ 34:30min it's on the fly : https://youtu.be/nFOSZyp4sw0?t=2098

kikislater, 2018-10-16:
Wrong previous input. OpenCv is used to defisheye pictures

Pix4dfields is similar to COLMAP + Aerial mapper ( https://github.com/ethz-asl/aerial_mapper )

It's mostly based on Eigen AKAZE feature tracking algorithm ( http://metrology.survice.com/sites/metrology.testing.survice.com/files/cmsc-16-initial.pdf ) and Ceres Solver (Dogleg trust region methods and SPARSE_NORMAL_CHOLESKY with the Tukey biweight loss function which aggressively which attempts to suppress large errors.).

About position of sensors : Green is the master sensor and position of each sensor is shift by theses datas =

[image]

smathermather, 2018-10-16:
@kikislater -- for COLMAP + Aerial mapper, you mean the homography approach at aerial_mapper?

kikislater, 2018-10-18:
Yes ! All the process except reflectance computation and pixel shifting from main sensor

LucasMM86, 2018-10-19:
Hello, I am thinking in scripting a bit cleaner the method I mention above. I want to do it in such a way that it is compatible with split merge. Do you thing this has sense? Is someone already rewriting split merge or working on that?

poynting, 2018-10-20:
Sorry to cross-post from related https://github.com/OpenDroneMap/OpenDroneMap/issues/190, but I'll drop this here as it seems this related thread is more active.

For RedEdge data, MicaSense offers some (python) code to get users started on radiometric processing, metadata extraction, band alignment, etc. There are examples there for automatically detecting reflectance panel images and using that calibration to convert flight images to reflectance, as well as general example describing the process of undistortion, alignment, and export of aligned images to RGB, CIR, or a multi-layer tiff while removing the unmatched overlap.  The same concepts should be applicable to most multispectral cameras.

https://github.com/micasense/imageprocessing
https://micasense.github.io/imageprocessing/Alignment.html

Full disclosure: I'm one of the authors and I work at MicaSense.

smathermather, 2018-10-24:
Thanks @poynting -- this will be a great addition.

poynting, 2018-10-24:
Right @BORDENEUVE, I wasn't suggesting that the rendered images be used (although with the example code, that could be easy to do, since ODM only supports JPEG).

One approach is to use a similar method of band alignment as outlined in our example code and then output a multi-band TIFF file with the radiometrically corrected aligned images.  There's code in the example (using gdal) to create that output. Next, ODM would read those TIFF files (Issue #190) and perform image-image alignment.   Some users of our code examples take this approach for research purposes where black-box photogrammetry products aren't an option.  

The ultimate output that I think most uses would be interested in would be a multi-band geotiff with all of the bands included that represents radiometrically-corrected data (either radiance or reflectance).

BORDENEUVE, 2018-10-24:
it is sure that a photo format geotiff mutibandes would be very good I see it with satelitte images and software like SNAP so you advocated that ODM also face the calculation of different indices. I think that a software like QGIS offers more possibility of working but it would be more difficult to manage this type of image it works very well using tape by tape (each of which is used as a raster layer) the problem being to know if the superposition and can be also the assembly of pictures is perfect.

MatthiasSiewert, 2019-05-14:
I haven't read the entire discussion, but here some thoughts:

Pix4D is calculating different vegetation indices such as NDVI at the end of the process. However, perfectly aligned / co-registered bands are necessary. 

The final caulculations could leverage grass gis. Specifically r.mapcalc:
https://grass.osgeo.org/grass74/manuals/imageryintro.html
https://grass.osgeo.org/grass74/manuals/r.mapcalc.html

From my outside perspective, I would suggest, that this is probably best served as a plugin similar to the contour plugin. The plugin could also include a module for radiometric calibration from reflectance targets.

newcomb-d, 2019-05-15:
See also GRASS GIS i.vi for more vegetation indices

https://grass.osgeo.org/grass76/manuals/i.vi.html

-- 
Doug Newcomb - Cartographer
USFWS
551F Pylon Dr
Raleigh, NC
919-856-4520 ext. 14 doug_newcomb@fws.gov
---------------------------------------------------------------------------------------------------------

*NOTE: This email correspondence and any attachments to and from this
sender is subject to the Freedom of Information Act (FOIA) and may be
disclosed to third parties.*​

BORDENEUVE, 2019-05-15:
by doing research on the web I could find formulas for the calculation of 8 crop indices (NDVI, ENDVI, CVI, EVI, MSR, OSAVI, LAI, GNDVI)
if it interests me some I could write them here

newcomb-d, 2019-05-15:
https://www.indexdatabase.de/db/i.php

-- 
Doug Newcomb - Cartographer
USFWS
551F Pylon Dr
Raleigh, NC
919-856-4520 ext. 14 doug_newcomb@fws.gov
---------------------------------------------------------------------------------------------------------

*NOTE: This email correspondence and any attachments to and from this
sender is subject to the Freedom of Information Act (FOIA) and may be
disclosed to third parties.*​

BORDENEUVE, 2019-05-15:
thanks for your list

BORDENEUVE, 2019-06-03:
here is a text that deals with the problem of multispectral images taken by drone
[mo2015-pub00047644.pdf](https://github.com/OpenDroneMap/ODM/files/3249766/mo2015-pub00047644.pdf)

merkato, 2019-07-20:
There's possibility for funding of this feature. 

Our friends from polish State Forestry are looking for multispectral images and vegetation indices in ODM/WebODM, as they have Sequoia cam, and declared financial support for developer, in crowdfunding model, or direct.

@pierotofy @smathermather @dakotabenjamin - first thing we must know, "how much?" Can you help?

pierotofy, 2019-07-21:
Hey @merkato :hand: that's great to hear!

I could have availability to help with this around the first weeks of September. Get in touch if you want to get the process started https://masseranolabs.com/contact

BORDENEUVE, 2019-07-30:
Hello
I do not know if this idea is good or not, but here it is rather than to assemble several bands of colors to make an image of it, why not do the opposite. We assemble the RGB photos and after we decompose it by photo processing in several images red, green, blue and infrared

tgmueller, 2019-12-29:
I'm wondering how things are going with the development of capabilities to import sequoia data into ODM.  The Tif file output is readable in real world coordinates by qgis or arcgis and I can't find utilities to do this.   But as I read all of this, it  looks like the images won't align with out radiometric correction.   It also seems like Micosense has some code that does the importing and correction but more work is needed before its ready for ODM.   Do I have all of this correct?

smathermather, 2019-12-30:
Hi Tom, ongoing work is being reported on here: https://community.opendronemap.org/t/tiff-multispectral-support/3232

tgmueller, 2020-01-04:
Thanks @smathermather

pierotofy, 2020-02-03:
We are now aligning multispectral cameras (and enforcing position contraints of the cameras during bundle adjustment in multi-camera setups as well).

There might still be some alignment issues due to inaccurate camera lens undistortion or due to texturing (try --texturing-data-term area instead of gmi if you see them).

I'll leave this open until we can confirm that alignments are good.

BORDENEUVE, 2020-02-06:
good evening pierotofy
I saw that ODM now had the possibility to work in multichannel and to perform the index calculation (NDVI SAVI and others). I have several questions on the export of these results, will be able to recover your orthomosaics of these indices
and will there be mono bands orthomosaics (red, green, blue, nir and other spectrum) or an orthomosaics with all the bands group together
awaiting your agreed responses my greetings

smathermather, 2020-02-11:
Hi @BORDENEUVE, can you post this query in community.opendronemap.org (if you haven't already). We try to keep these questions in that forum where they are more readily find-able than in issues.

Thanks!

BORDENEUVE, 2020-02-11:
good evening smathermather
following your answer I registered on the forum opendronemap.org and I will ask my question
thank you for everything

smathermather, 2020-04-14:
Closed as of https://github.com/OpenDroneMap/ODM/pull/1057!

closed 2020-04-14 by smathermather