OpenDroneMap/ODM #503: not an issue, only questions

LucasMM86, 2017-03-02:
Hello,
¿how can I know which version of ODM am I using? (V0.2-RC1 for example)
¿a simple way to update to the last version?
Thanks, Lucas

LucasMM86, 2017-03-02:
I use ubuntu 14.04 on a VM.

dakotabenjamin, 2017-03-02:
I've haven't addressed the issue of updating yet, so I don't have a good answer for you. If you know a good way to keep track of that, we can start to integrate for future versions.

LucasMM86, 2017-03-02:
I think that a --version flag would be great.
but maybe this is simpler: updating the line "Current version: 0.2 (this software is in beta)" in readme.md fo r each version?

dakotabenjamin, 2017-03-02:
Both methods probably will work. I'll see if I can get this implemented by v0.3.

dakotabenjamin, 2017-04-03:
this has been added https://github.com/OpenDroneMap/OpenDroneMap/commit/cee7b633123e114aa1f5635680cb7d9df592d924

closed 2017-04-03 by dakotabenjamin