OpenDroneMap/ODM #508: Add ID number to GCP format

smathermather, 2017-03-08:


dakotabenjamin, 2017-03-08:
Combine with #507 and awaited GCP updates. Also we should work on the GCPoint object.

yjmenezes, 2017-03-09:
i think, it would be good to have or a flag (On/Off or XY,Z,XYZ ) or standard deviation for XY and Z. This way we could turn points ON/OFF by flag or setting point weight.

column 1 # signal to turn a point into a comment, or to insert comments inside the GCP file.

dakotabenjamin, 2017-03-10:
I like the idea of adding deviations or other error values, but I'm not sure on the implementation.

yjmenezes, 2017-03-11:
@dakotabenjamin 

I hope  it will be not so hard to implement.
The weight matrix W can be a diagonal matrix derived from point standard deviation.

Please, take a look in your spare time. Chapter4 and 9

www.mygeodesy.id.au/documents/Chapter%204.pdf
www.mygeodesy.id.au/documents/Chapter%209.pdf

I was looking GRASS but no weight is used, just a status to turn GCP On/Off:
https://grass.osgeo.org/grass72/manuals/addons/v.ply.rectify.html

yjmenezes, 2017-04-16:
@dakotabenjamin , @smathermather 

please, take a look, an alternative approach for affine3D. Just an idea.
i wrote a shell script and an octave script;
octave path must be edited inside shell script.

http://cartognu.org/odm
odm_affine3d_2017-04-16.tar.bz2

thx

yjmenezes, 2017-04-22:
@dakotabenjamin 

thank you.

Please, if you have a free time, i process boruzsyn with alternative approach for affine3D.
cartognu.org/odm/odm_boruzsyn_affine3d_2017-04-22.tar.bz2
take a look on HV04 screen shots and checksol.txt  pto ( 10,11,12  ).

cheers

dakotabenjamin, 2017-04-24:
When I get a chance, I will do that. It looks very interesting, but I'll have to read up a bit.

closed 2018-03-29 by dakotabenjamin