OpenDroneMap/ODM #512: Exception: Child returned 137 while processing images

renandp11, 2017-03-14:
Hello everyone!

I was trying to process my first set of images but process ended in exception. I was running fine but errored out, any advice? 

Error Log:

2017-03-14 03:06:16,738 -------------------------------------------------------
2017-03-14 03:06:16,842 DJI_0041.JPG resection inliers: 1351 / 1353
2017-03-14 03:06:16,874 Adding DJI_0041.JPG to the reconstruction
2017-03-14 03:06:33,085 Ceres Solver Report: Iterations: 2, Initial cost: 1.081015e+03, Final cost: 1.077634e+03, Termination: CONVERGENCE
2017-03-14 03:06:33,644 Bundle setup/run/teardown 2.96273708344/12.3001670837/0.562075853348
2017-03-14 03:06:33,767 Removed outliers: 0
2017-03-14 03:06:35,217 -------------------------------------------------------
2017-03-14 03:06:35,266 DJI_0047.JPG resection inliers: 1348 / 1358
2017-03-14 03:06:35,296 Adding DJI_0047.JPG to the reconstruction
2017-03-14 03:06:48,752 Ceres Solver Report: Iterations: 1, Initial cost: 1.084514e+03, Final cost: 1.082147e+03, Termination: CONVERGENCE
2017-03-14 03:06:49,323 Bundle setup/run/teardown 3.03747391701/10.2676420212/0.570981025696
2017-03-14 03:06:49,440 Removed outliers: 0
2017-03-14 03:06:50,785 -------------------------------------------------------
2017-03-14 03:06:50,829 DJI_0330.JPG resection inliers: 1259 / 1299
2017-03-14 03:06:50,853 Adding DJI_0330.JPG to the reconstruction
/odm_app/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all: line 11:  4397 Killed                  $DIR/opensfm reconstruct $1
Traceback (most recent call last):
  File "/odm_app/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/odm_app/OpenDroneMap/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/odm_app/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
vagrant@vagrant-ubuntu-trusty-64:/odm_app/OpenDroneMap$

pierotofy, 2017-03-14:
The keyword to notice in this log is `Killed`. You probably haven't allocated enough RAM to your vagrant VM to process a dataset of that size.

renandp11, 2017-03-14:
Thanks a lot, I'll check RAM settings for VM

renandp11, 2017-03-16:
Hi pierotofy the RAM issue is fixed, now I found another one when Mergin depthmaps, could this be a machine resources also? (I'm processing less images)

2017-03-16 02:28:09,583 Merging depthmaps
Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/odm_app/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/odm_app/OpenDroneMap/scripts/opensfm.py", line 90, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/odm_app/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

pierotofy, 2017-03-16:
This looks like the same issue as #509

renandp11, 2017-03-16:
Hi, I've just tried to process fewer images and everything is done now, geotiff and mesh ... really cool! I think its a matter of resources on my virtual machine. I'm running vagrant on virtualbox for win 10 
Thanks a lot for your support

closed 2017-04-03 by dakotabenjamin

marvinlange, 2017-04-13:
Is this the same issue ?

> Segmentation fault
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 113, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

>

renandp11, 2017-04-13:
Hi marvinlange
Not sure if issue is the same, I've troubleshooted it trying to process fewer pictures, like 10 pics, then reconstruction was successful, so it was a matter of pc resources when tried to process more than 25 pics ... would you try fewer pictures?