OpenDroneMap/ODM #514: what the different between two obj files

assbben, 2017-03-18:
I find an interesting thing 
[image]

This model is created by openmvg and openmvs, when i use odm_orthophototo make orthophoto it always report 'segmentation fault (core dump)'

[image]

After i made a x-zmirrored model by blender,it can work .
But the orthophoto is not good

[image]

here are my question 
1.why segmentation fault did`nt happened when i used changed model 
2.i find it has the same change in odm after using odm_georeferencing module
I think its very interesting and i want to know what`s the different inside and what odm_georeferencing had done.
Thank you

dakotabenjamin, 2017-03-22:
Hi, Can you post the specific stdout around the segfault? That would be very helpful. Also I would love to test the data so if you post it I will try to reproduce the error.

dakotabenjamin, 2017-03-22:
lastly I don't think I can provide much support if you are not using ODM to build the 3d models- I don't have the experience with openmvg.

assbben, 2017-03-24:
do you mean what core file include? or my command to use orthophoto modules?
my texture is too big(90mb), i can't post it to github.

dakotabenjamin, 2017-03-24:
If you can post to Dropbox or another filesharing service that would be helpful. If you are able to share the images and tell me what commands you ran, then I can try to reproduce.

assbben, 2017-03-25:
https://www.dropbox.com/s/24gdgozhdjdyc4x/my_modle.tar.gz?dl=0

dakotabenjamin, 2017-03-27:
I think I'm beginning to understand the problem. You cannot reasonably make an orthophoto without translating the model to georeferenced coordinates. This is what odm_georeferencing does. With either a [ground coordinates points file](https://github.com/OpenDroneMap/OpenDroneMap/wiki/Running-OpenDroneMap#running-odm-with-ground-control), or coordinates in the photo's EXIF data, it extracts a set of coordinates for each camera then connects those to the same cameras on the bundle file (which is not referenced).

assbben, 2017-03-27:
Thank you! Its the problem of negative vector?  I think opencv can't render pixel to negitive position in `renderPixel`function.That`s why segment fault happened. I've used odm_georeferencing to make translation,but openmvg and openmvs bundle file format is different to odm and i'll try it again.

dakotabenjamin, 2017-03-27:
That would make sense as to why the x-z mirrored model worked partially.

dakotabenjamin, 2017-04-03:
If you have any more questions, feel free to reopen.

assbben, 2017-04-08:
Well, i've finally installed the lastest version successfully.And what is include in bundle file ,point cloud or others? When i open bundle_r000.out it has many data. I'll use openmvg again to find how to make bundle file. Besides i find data in coords.txt is camera points in my model coords, I can get those points when using openmvg. maybe gcp file can be instead by  coords file. Because gcp file needs pixel data.

assbben, 2017-04-10:
After I got bundle file,I test odm_georef . It reports "Segmentation fault (core dumped)" . But when I used a smaller obj file, it does report error.My new question is whether georef module has the limit of obj ｓｉｚｅ．
PS: Can you reopen this issue or I need to open a new  issue? @dakotabenjamin

dakotabenjamin, 2017-04-10:
Can you post the error?

dakotabenjamin, 2017-04-10:
Where exactly the segfault happened i mean

assbben, 2017-04-10:
Thank you !
this is my command "./odm_georef -bundleFile '/home/bobo/test_openmvg/global/PMVS/bundle.rd.out' -inputCoordFile '/home/bobo/newodm_17-3-28/test/odm_georeferencing/coords.txt' -inputFile '/home/bobo/Desktop/test_mvs/test16/mesh16.obj'    -outputFile '/home/bobo/Desktop/test_mvs/test16/geo.obj' 
Segmentation fault (core dumped)
" 
This core file
＂Core was generated by `./odm_georef -bundleFile /home/bobo/test_openmvg/global/PMVS/bundle.rd.out -inp'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00000000005142b2 in ?? ()＂

Maybe the error is not because of the limit of modle.And  I'll test again  tomorrow.

dakotabenjamin, 2017-04-10:
OK so like I said before I can't help much since the model was not generated in ODM; I just don't know what could possibly be wrong since OpenMVG is not our project. If you get the same results tomorrow let me know.

assbben, 2017-04-11:
＂
[New Thread 0x7fffde410700 (LWP 12937)]
[New Thread 0x7fffddc0f700 (LWP 12938)]
[New Thread 0x7fffdd40e700 (LWP 12939)]
[New Thread 0x7fffdcc0d700 (LWP 12940)]
[Thread 0x7fffdd40e700 (LWP 12939) exited]
[Thread 0x7fffdcc0d700 (LWP 12940) exited]
[Thread 0x7fffddc0f700 (LWP 12938) exited]
[Thread 0x7fffde410700 (LWP 12937) exited]

Thread 1 "odm_georef" received signal SIGSEGV, Segmentation fault.
0x00000000005142b2 in Eigen::DenseStorage<float, 2, 2, 1, 0>::DenseStorage(Eigen::DenseStorage<float, 2, 2, 1, 0> const&) ()
＂
this is report in gdb  
And I think openmvg is similar to opensfm, because It must use pmvs or other tools to out put dense point cloud.

dakotabenjamin, 2017-04-11:
I think that because odm_georef was built specifically for the output produced by ODM and not for general use, you are running into a specific bug or issue that it doesn't account for. This is not something we currently have a reason to fix due to lack of resources. Eventually we will probably want to generalize odm_georef and all our modules for a more broad usage but for now that is not possible. If you have c++ knowledge and want to track down the issue, I will gladly test a pull request as well. 

Can you post the coords, bundle, etc. files? I'll try to replicate.

assbben, 2017-04-11:
https://www.dropbox.com/s/6qg620xh2qh3u0k/17.4.11_georef_test.tar.gz?dl=0

Now I'm using pmvs to output dense point cloud to test whether error is caused by different mvs tools.
Can you give me some paper about georefence and orthophoto,and i'll try to make it .

closed 2018-03-28 by dakotabenjamin