OpenDroneMap/ODM #515: Checking if the process is running.

sensehawk, 2017-03-19:
I have been trying to process 200 images on a 8 cores/32 GB machine. I am unable to process to create all the outputs. It creates the opensfm folder and creates all files/folders and then the CPU goes to 0. Although the task is still running, output folders are not getting created. I have waited for 8 hours before killing the process. I have tried --cmvs-maxImages 100 to decrease memory usage as I saw earlier that RAM was filling up to max. 

I tried to process 50 images and that created all the outputs on the same machine.  Can some one help me understand what is happening please.

smathermather, 2017-03-19:
A few questions/suggestions:

Do the images have geolocation in the exif file?

CMVS flags may not help here. I would instead set the -opensfm-processes to 4 as a start.

rohitrohu, 2017-03-19:
Images are geo-referenced. 
Let me try with -opensfm-processes set to 4. 
Will let you know once i get some results. Thanks for the suggestion!

rohitrohu, 2017-03-20:
This is on a 60GB RAM/8 Core machine.

 File "/home/ubuntu/OpenDroneMap-master/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap-master/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/ubuntu/OpenDroneMap-master/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132

I ran it with -opensfm-processes set to 6 and --cmvs-maxImages 100. Any idea what I can do to make it work.

smathermather, 2017-03-20:
I wouldn't set the cmvs settings at all. @dakotabenjamin -- what happens when PMVS and OpenSfM settings are used together?

rohitrohu, 2017-03-20:
Okay. I am still trying to get my head around the whole process and the run time variables.  How do commercial software manage resources so efficiently. I think there should be a way to set these critical parameters automatically depending on the configurations, right?

smathermather, 2017-03-20:
Lots of stuff in the pipeline for doing things more efficiently: https://smathermather.com/2017/03/08/scaling-opendronemap-necessary-and-fun-next-steps/

pierotofy, 2017-03-20:
@rohitrohu Out of curiosity... if you use the docker method to run ODM, does it work?

rohitrohu, 2017-03-21:
@smathermather That is interesting! I hope we get there soon. Meanwhile, is there a way to get this working? I tried with --OpenSfM 6 on a 8 core/60GB machine. No luck. I can share machine details if needed to understand this issue. 

@pierotofy Haven't used Docker. I am running it directly on a Ubuntu machine on AWS.

rohitrohu, 2017-03-21:
After setting only OpenSFM process to 7 on a 8 core machine.

`2017-03-21 06:42:15,981 Common tracks: 3844
/home/ubuntu/OpenDroneMap-master/SuperBuild/src/opensfm/bin/run_all: line 11:  1736 Illegal instruction     (core dumped) $DIR/opensfm reconstruct $1
Traceback (most recent call last):
  File "/home/ubuntu/OpenDroneMap-master/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/OpenDroneMap-master/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/ubuntu/OpenDroneMap-master/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 132

[1]+  Exit 1                  sudo ./run.sh --project-path /home/ubuntu/projects/asifabad/ --opensfm-processes 7`

smathermather, 2017-03-21:
I can't see why this is failing. Is this a dataset you can share for testing?

rohitrameshrao, 2017-03-22:
I can share the data set or SSH access to the AWS machine on which I am trying to process. I should have mentioned this earlier, there are a total of 590 images (4.8 GB) at 20 MP. Is that a large data set to process?

dakotabenjamin, 2017-03-22:
It's definitely on the larger side, but doable for sure. 

It may be an OpenSfM installation issue. Can you `cd SuperBuild/src/opensfm` and run `python setup.py` ?

rohitrameshrao, 2017-03-22:
Gives me an error.

[image]

dakotabenjamin, 2017-03-22:
Odd. Try deleting `opensfm/build` and `opensfm/cmake_build` before rebuilding using `python setup.py build` to make sure that anything that you've changed outside opensfm gets properly linked.

rohitrameshrao, 2017-03-23:
Tried again. Not sure if it successfully built it. 

[image]

I was able to process around 200 images data set on the same machine. Without opensfm, I could not have got the outputs right?

dakotabenjamin, 2017-03-23:
You're probably right. I would try either a clean install of ODM or using the docker build on opendronemap/opendronemap:latest

rohitrameshrao, 2017-03-27:
Still unable to process it. Here is the link to data set incase some one can try processing at their end.
https://s3-ap-southeast-1.amazonaws.com/sensehawk-public/odm_stockpile_example.zip

dakotabenjamin, 2017-03-27:
I'll download it overnight and check it out in the morning (GMT-0400)

rohitrameshrao, 2017-03-28:
@dakotabenjamin Thanks! Do let me know if you are able to process it!

dakotabenjamin, 2017-03-29:
So I was able to process through texturing so far (still running) with no issue on a clean pull from `master`. I cannot explain the `Illegal instruction` issue, but I think this is not a problem with ODM. 

Do you have OpenCV installed somewhere else on your system?

rohitrameshrao, 2017-03-30:
Oh okay. Strange. I took a new machine, installed Ubuntu and then ODM. Did not install OpenCV separately. 

Any specific parameters you put in that I can try here?

szimnave, 2017-04-01:
hello guys, I have the same problem here..ODM stuck in the middle of processing photos. I have 311 photos. please help. thank you

rohitrameshrao, 2017-04-02:
Update: I took a machine with 125 GB RAM & 16 cores and tried processing on it. It still doesn't process.  I see OpenSFM and PMVS folders and a few outputs inside OpenSFM, nothing inside PMVS.

dakotabenjamin, 2017-04-03:
@szimnave can you copy-paste the command you used to run, the error you are receiving, and the specs of your machine?

@rohitrameshrao to be clear, which error are you getting this time?

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/checking-if-the-process-is-running/399

closed 2018-03-28 by dakotabenjamin