OpenDroneMap/ODM #516: Documentation Mega-Issue

dakotabenjamin, 2017-03-22:
Use this for suggestions, ideas, complaints, etc.

pierotofy, 2017-03-22:
I'd be happy to see more documentation on how to get hacking with ODM (and in particular to understand the ecto cells). But this is a good list!

rohitrameshrao, 2017-03-23:
It would be helpful if we can add some information about hardware usage and recommended hardware.

smathermather, 2017-03-23:
Good point. This is a common question, and I have few concrete answers for people.

hblanken, 2017-03-26:
Would be great to add best practice mission and image resolution/settings together with a list of camera types with a link to an inventory of matrix.txt and distortion.txt files for each camera type. Clearly there will be must do's and but also recommendations which will vary by use case objective.

smathermather, 2017-03-26:
Good call, @hblanken.

rohitrameshrao, 2017-03-27:
I think we also need more information on accuracy. Very critical to know how accurate the reconstruction is.

dakotabenjamin, 2017-03-27:
@rohitrameshrao that is a feature currently in the works, hopefully soon.

pepperlk, 2017-04-12:
Gonna give some performance input on a few runs of ODM:
Using the odm_data_toledo with a big VM (16 cores 32gb ram) I clocked in at 22 min (avg) across 5 runs
Trying to understand the hardware to processing time ratio..  
Ill post more updates as I have time to run tests but I think the performance is very good for this project seeing that mark.. 
That vm runs at $0.98 hr so decent bang for the literal buck..

hblanken, 2017-05-02:
Building on @rohitrameshrao and @dakotabenjamin and @pepperlk comments - It would be really good to build a list or table or other graphical representation of the tradeoff decisions to make to achieve quality/accuracy vs cost/time. I would suggest the quality depends on its horizontal and vertical accuracy. The absolute vertical accuracy of a ODM asset could be measured as the average discrepancy between sample points of ODM and surveyed positions (considered as the reference). Following the theory of normal distribution of errors, the accuracy can be stated at the 1-sigma (68% confidence interval) or 2-sigma (95% confidence) interval.

dakotabenjamin, 2018-03-21:
All: docs.opendronemap.org is now live!

hblanken, 2018-03-23:
Fantastic read - I saw the 'flying tips' section is empty - let me contribute via this comment and make a start:

```
Please do a final check to ensure a great outcome before you fly and before you start processing:

[ ] Do at least 2 rounds 'Point of Interest' scenes for 3D objects: We recommend to shoot at 5 m/s rotation, shoot one image every 3 sec, resolution at 2.7k is great, 4k is often not needed.
[ ] Ensure the images are all geotagged.
[ ] Add a 'Grid' shot for large areas : We recommend shoot with at least 60% overlap. A rule of thumb at 100m height is one shot every 3 seconds, with 20m/s.
[ ] Images stitch only if there is no sky in the background! So keep at least 45 degree camera angle.
[ ] We recommend 50-500 images, this is often sufficient for most models.

Think of it as a spray can, the more you spray !equally! across your scenery, without gaps, the better the results.
``` 
I will think of more to fill the blanks

smathermather, 2019-11-14:
Docs have been substantially written. There's always more to do, but we should refresh with a new issue, as much of this has been addressed or is otherwise no longer an issue.

closed 2019-11-14 by smathermather