OpenDroneMap/ODM #518: Reconstructing a scene using a subset of test images

araza7865, 2017-03-23:
I am currently working on themes related to 3D reconstruction from aerial images for which I am using OpenDroneMap. Currently I am working on designing the optimal trajectory for a drone such that as much information about a scene is gathered as possible while using a limited number of drones at any given time. To this end, I am using the data set "Bellus". As a preliminary step, I tried reconstructing a scene using only 4 photos from the data set. While the OpenDroneMap reconstructs the entire scene using the given 122 images, I get an exception when I run it for 4 images. The exception is:

Traceback (most recent call last):
  File "/home/ahsan/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/ahsan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/home/ahsan/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/home/ahsan/OpenDroneMap/project2/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "/home/ahsan/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/ahsan/OpenDroneMap/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/ahsan/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

Could you please tell me what the issue might be, or is it even possible to use a subset of a data to construct an imperfect scene for testing purposes?

Thanks

dakotabenjamin, 2017-03-23:
If you want a very small subset, you may need more than 4 photos. In general for each feature you want 3 photos to match with in order to build a reconstruction. I would try to make it so there are 3 overlapping photos in a sequence which each have 2 or more overlapping photos overlapping them. That would make at least 9 photos. Maybe these will work:
```
wkt_geom	filepath	filename	longitude	latitude	altitude	north	azimuth	gps_date	img_date
Point (-81.70353190000000154 41.2257087999999996)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1365_RGB.jpg	IMG_1365_RGB.jpg	-81.7035319	41.2257088					2015:07:31 14:31:53
Point (-81.70394579999999962 41.22585790000000117)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1366_RGB.jpg	IMG_1366_RGB.jpg	-81.7039458	41.2258579					2015:07:31 14:31:58
Point (-81.70308430000000044 41.22566170000000341)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1354_RGB.jpg	IMG_1354_RGB.jpg	-81.7030843	41.2256617					2015:07:31 14:30:14
Point (-81.70348500000000058 41.22588760000000008)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1304_RGB.jpg	IMG_1304_RGB.jpg	-81.703485	41.2258876					2015:07:31 14:25:43
Point (-81.70331290000000024 41.22535140000000098)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1355_RGB.jpg	IMG_1355_RGB.jpg	-81.7033129	41.2253514					2015:07:31 14:30:18
Point (-81.70345000000000368 41.22502899999999926)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1356_RGB.jpg	IMG_1356_RGB.jpg	-81.70345	41.225029					2015:07:31 14:30:24
Point (-81.70368410000000381 41.22537140000000022)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1409_RGB.jpg	IMG_1409_RGB.jpg	-81.7036841	41.2253714					2015:07:31 14:35:55
Point (-81.70411900000000571 41.22549850000000049)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1410_RGB.jpg	IMG_1410_RGB.jpg	-81.704119	41.2254985					2015:07:31 14:36:00
Point (-81.70387920000000292 41.22605899999999934)	W:\Aircraft\data\2015-07-31-BellusRdField\odm_data_bellus\images\IMG_1346_RGB.jpg	IMG_1346_RGB.jpg	-81.7038792	41.226059					2015:07:31 14:29:31
```

araza7865, 2017-03-23:
Great, thanks, will try it!

araza7865, 2017-03-24:
It worked, thanks dakotabenjamin!

dakotabenjamin, 2017-03-24:
Good to know, let us know on [gitter](https://gitter.im/OpenDroneMap/OpenDroneMap) how your project goes!

closed 2017-03-24 by dakotabenjamin