OpenDroneMap/ODM #521: 2.5D Mesh Generation

pierotofy, 2017-03-25:
So I've been meaning to do some work on improving the orthophoto results (I've started a conversation here time ago: https://github.com/OpenDroneMap/OpenDroneMap/pull/444) and I think I can implement it.

Achieving a 2.5D mesh can be done using a constrained delaunay triangulation and it's detailed here https://github.com/CloudCompare/CloudCompare/blob/579ee91594e16beba5863e2e4d15cf0729eb68a8/libs/qCC_db/ccMesh.cpp and here https://github.com/CloudCompare/CloudCompare/blob/579ee91594e16beba5863e2e4d15cf0729eb68a8/CC/src/Delaunay2dMesh.cpp

[image]

CloudCompare offers this feature built-in and has a command line mode http://www.cloudcompare.org/doc/wiki/index.php?title=Command_line_mode, however I don't think I'd want to include all of its dependencies (including QT!) just for the sake of generating a 2.5D mesh.

I could isolate just the code that takes care of the 2.5D mesh generation (and add some filtering of my own to reduce the spikes), but it will be necessary to add http://www.cgal.org/ as a dependency. PCL does not have any support for delaunay triangulations.

@dakotabenjamin what are your thoughts?

pierotofy, 2017-03-25:
Also cc @smathermather

smathermather, 2017-03-25:
Can we also isolate just the bits of CGAL we need, or is that a fools errand?

pierotofy, 2017-03-25:
CGAL is complicated to just use as-is, I wouldn't venture as far as isolating a single component. What are the downsides of including CGAL, besides increased compilation time and docker image size?

smathermather, 2017-03-25:
Just that. We can include it and then put it on a list of dependencies to remove later. I'm happy to take that approach: we're at an add features stage for the project, so I'd say proceed!

pierotofy, 2017-03-25:
Feel free to assign me the task.

dakotabenjamin, 2017-03-27:
Can we migrate all the current PCL functions over to CGAL? I don't have any experience with it. Removing PCL could speed up complication times significantly.

pierotofy, 2017-03-27:
Probably. The good news is that there's a prebuilt package for including CGAL in Ubuntu. `apt-get install libcgal-dev`.

dakotabenjamin, 2017-03-27:
That is pretty good news. PCL is heavy.

I would suggest for now just getting the 2.5D work in and we can sit down and sort through the possibilities of a full migration at a later date.

pierotofy, 2017-03-29:
Boom! 

[image]

Still need to do some smoothing / filtering.

pierotofy, 2017-03-29:
[image]

smathermather, 2017-03-29:
Awesome!

pierotofy, 2017-04-03:
Starting to look a lot better:

[image]

pierotofy, 2017-04-05:
Still need to do further smoothing, but I think we're going to get free cropping along with this enhancement!

[image]

This is due to the delaunay triangulation and outlier removal which makes some nice sharp edges around the resulting orthophoto (vs poisson).

kikislater, 2017-04-06:
It seems nice, how to test it ?

pierotofy, 2017-04-06:
Checkout this commit https://github.com/pierotofy/OpenDroneMap/commit/503671487020fa4ff6dad77bdf984bb5592f5313

And run with the `--use-25dmesh` argument. It's still a work in progress though, expect improvements in the next few days.

pierotofy, 2017-04-06:
You will also need to compile the odm_25dmeshing module and copy the resulting binary in the bin folder. If you are not comfortable with cmake, make, etc. I would wait until I configure the configure.sh script properly.

dakotabenjamin, 2017-04-06:
How is the cropping? Any areas that are over-cropped when compared to a ortho generated via 3d mesh?

pierotofy, 2017-04-06:
It depends on the outlier removal parameter. The more outliers are removed, the more conservative the crop is.

kikislater, 2017-04-07:
Ok, I'm familiar with make and cmake but use it with docker under archlinux. I will wait to test.
Control of outlier is planned ?

pierotofy, 2017-04-07:
Yes, I'm adding a outlier removal parameter that can be tuned as well as a max vertex count parameter that defines the precision of the output mesh.

pierotofy, 2017-04-07:
Old:

[image]

2.5D:

[image]

kikislater, 2017-04-07:
Look nice!
Any idea why struture between 2 trees is wavy with delaunay ?

pierotofy, 2017-04-07:
I'm not entirely sure; I'm not very familiar with the mvs texturing package, which is the component that overlays the texture over the surface model.

pierotofy, 2017-04-07:
Probably because of noise: 
[image]

kikislater, 2017-04-07:
Looks like more a shiny surface with different shine on each picture.

pierotofy, 2017-04-07:
No that's because illumination is turned on.

[image]

pierotofy, 2017-04-07:
Re-ran it with half the number of vertices (less noise, but less definition on edges):

[image]

[image]

pierotofy, 2017-04-07:
@kikislater you should be able to test the 2.5D mesh feature by checking out the changes in #542 and running `. configure.sh`.

closed 2017-06-14 by dakotabenjamin