OpenDroneMap/ODM #524: Problem with reconstruction.json

Davidlapalma, 2017-03-29:
Hello, my problem is that odm not find the reconstruction.json file due to this wasn't create on project folder. The most strange thing is that I was testing with other images and work perfectly. Please, can someone tell me what is the error? 
The message on the end is the next:
File "/.../OpenDroneMap/opendm/system.py", line 28, in run     raise Exception("Child returned {}".format(retcode)) Exception: Child returned 1

dakotabenjamin, 2017-03-29:
If other images worked fine, then it is probably an issue with your dataset, either the overlap is too low to create a reconstruction or some other reason. You can try to overcome this by increasing `--min-num-features` to something like 8000 or 10000

Davidlapalma, 2017-03-29:
And how i increase the min-num-features?

dakotabenjamin, 2017-03-29:
Check the [usage guide](https://github.com/OpenDroneMap/OpenDroneMap/wiki/Run-Time-Parameters) for how to run ODM with parameters.

Davidlapalma, 2017-03-29:
Ok thanks

Davidlapalma, 2017-03-29:
Ok thanks

Davidlapalma, 2017-03-30:
i change the min-num-features but that is not the problem
I send you 2 images of the collection.
if you see the problem please I would apreciate it.
Also, I work in Photoshop and the image overlap good
Please if it's possible,test my images 
[image]
[image]

dakotabenjamin, 2017-03-31:
I can't tell anything wrong with those two images. Can you post more of the error output?

Davidlapalma, 2017-04-03:
2017-04-03 11:29:29,747 Extracting focal lengths for image DJI_0211.JPG
2017-04-03 11:29:29,799 Extracting focal lengths for image DJI_0212.JPG
2017-04-03 11:29:30,189 Extracting ROOT_HAHOG features for image DJI_0211.JPG
2017-04-03 11:29:30,189 Extracting ROOT_HAHOG features for image DJI_0212.JPG
2017-04-03 11:29:32,471 Found 5659 points in 2.18905711174s
2017-04-03 11:29:32,476 Found 5673 points in 2.19341802597s
2017-04-03 11:29:33,340 Matching 1 image pairs
2017-04-03 11:29:33,359 Matching DJI_0212.JPG  -  1 / 2
2017-04-03 11:29:33,359 Matching DJI_0211.JPG  -  2 / 2
2017-04-03 11:29:34,573 DJI_0211.JPG - DJI_0212.JPG has 2098 candidate matches
2017-04-03 11:29:34,574 Robust matching time : 0.000715970993042s
2017-04-03 11:29:34,574 Full matching 2084 / 2098, time: 1.21415305138s
2017-04-03 11:29:35,005 reading features
2017-04-03 11:29:35,056 Merging features onto tracks
2017-04-03 11:29:35,079 Good tracks: 2084
2017-04-03 11:29:35,517 0 partial reconstructions in total.
Traceback (most recent call last):
  File "/home/.../OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/.../OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/home/parallels/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/home/.../OpenDroneMap/project/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/.../OpenDroneMap/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/.../OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

Davidlapalma, 2017-04-03:
and the exifs seems fine
DJI_0211
{
    "width": 2400, 
    "camera": "v2 dji fc330 1800 2400 perspective 0.5555", 
    "projection_type": "perspective", 
    "orientation": 1, 
    "focal_ratio": 0.5555555555555556, 
    "make": "DJI", 
    "gps": {
        "latitude": 28.071327166666666, 
        "altitude": 255.494, 
        "longitude": -15.457539749999999
    }, 
    "model": "FC330", 
    "capture_time": 1490350395.0, 
    "height": 1800
}

DJI_'212
{
    "width": 2400, 
    "camera": "v2 dji fc330 1800 2400 perspective 0.5555", 
    "projection_type": "perspective", 
    "orientation": 1, 
    "focal_ratio": 0.5555555555555556, 
    "make": "DJI", 
    "gps": {
        "latitude": 28.07138925, 
        "altitude": 255.494, 
        "longitude": -15.457602111111111
    }, 
    "model": "FC330", 
    "capture_time": 1490350396.0, 
    "height": 1800
}

dakotabenjamin, 2017-04-03:
There's no way to only match 2 images. You'll need 9 at the minimum.

Davidlapalma, 2017-04-03:
I have tried it with nora than 9 images but I have had the same results

dakotabenjamin, 2017-04-03:
So the real error is here:
```
IOError: [Errno 2] No such file or directory: '/home/.../OpenDroneMap/project/opensfm/reconstruction.json'`
```
This means that the software failed to create a reconstruction from the image matches. This is usually due to poor overlap, but that looks ok. My guess now is that there's some image distortion going on with your camera. This is the DJI FC330 which is found on the Phantom 4, no? We have lots of trouble with the DJI cameras because they have significant lens distortion. I have a script that can remove that: https://github.com/dakotabenjamin/CameraCalibration

Davidlapalma, 2017-04-03:
i will pass all the images of the folder

Davidlapalma, 2017-04-03:
https://www.dropbox.com/sh/nsq2tql1uxm3rzg/AAAM3-yH4RUa1U64Q0VJJ96va?dl=0

pierotofy, 2017-04-03:
My guess is that your pictures are too overexposed, and the feature matching algorithm is unable to find enough features.

[image]

Stretching the levels on each image might yield better results, but honestly, it's not going to look good:

[image]

I would recommend checking the settings on your DJI and shoot it again if you can with auto exposure settings.

Davidlapalma, 2017-04-04:
ok thanks, i will try it again

dakotabenjamin, 2017-04-04:
Thanks @pierotofy

theriny, 2018-01-31:
@dakotabenjamin I am getting the same error message

> IOError: [Errno 2] No such file or directory: '/home/.../OpenDroneMap/project/opensfm/reconstruction.json'`

I looked at my 'exifs' and noticed that the perspective for each file is '0' as seen in the 2 examples below:

{
    "width": 5184, 
    "camera": "v2 canon ef-s18-55mm f/3.5-5.6 is stm 5184 3456 perspective 0", 
    "projection_type": "perspective", 
    "orientation": 1, 
    "focal_ratio": 0, 
    "make": "Canon", 
    "gps": {}, 
    "model": "EF-S18-55mm f/3.5-5.6 IS STM", 
    "capture_time": 1472839245.015, 
    "height": 3456
}

{
    "width": 5184, 
    "camera": "v2 canon ef-s18-55mm f/3.5-5.6 is stm 5184 3456 perspective 0", 
    "projection_type": "perspective", 
    "orientation": 1, 
    "focal_ratio": 0, 
    "make": "Canon", 
    "gps": {}, 
    "model": "EF-S18-55mm f/3.5-5.6 IS STM", 
    "capture_time": 1472839251.009, 
    "height": 3456
}

Probably important to mention: The images I am processing were captured with a stationary camera (I know that defeats the purpose of structure from 'motion'). However, the object of interest is being rotated about the z-axis in front of a solid blue background in between captures (I have 15 total images). Could the fact that the camera being stationary be the reason my perspective is returning '0' in the 'exif' files? Could the inability to create the 'reconstruction.json' be related to there not being movement of the camera even the object of interest is being rotated?

smathermather, 2018-02-11:
@theriny -- rotating the object is an anti-pattern for most structure from motion approaches, including the one used by OpenDroneMap. There are a variety of problems with this approach, including that the background doesn't change and thus overwhelms the signal of the object of interest.

dakotabenjamin, 2018-03-17:
This has been solved through the implementation of flat plane initialization

closed 2018-03-17 by dakotabenjamin