OpenDroneMap/ODM #527: Add --build-overviews option

smathermather, 2017-03-30:
Add --build-overviews option per https://github.com/OpenDroneMap/WebODM/issues/127 to allow for the building of overviews via gdal. It probably isn't something that should be enabled by default, but useful option, and might be a good default for webodm.

dakotabenjamin, 2017-03-31:
I'll add this to #525

dakotabenjamin, 2017-03-31:
This is a bit of a difficult problem because I'm not sure how to set up the `gdaladdo` command call. Let's look at the [usage page](http://www.gdal.org/gdaladdo.html)
```
gdaladdo [-r {nearest,average,gauss,cubic,cubicspline,lanczos,average_mp,average_magphase,mode}]
         [-b band]*
         [-ro] [-clean] [-oo NAME=VALUE]* [--help-general] filename levels
```

- `-r`: Do we assume a default here or expose as a parameter? 
- `-b`: ignore? Does not selecting a band assume all bands?
- `-ro`: Internal? External? 
- `-clean`: ignore
- `-oo`: ?
- `filename`: easy, just pass `context.odm_orthophoto_tif`
- `levels`: Should we allow adjusting these, or will 2 4 8 16... work?

What about external GeoTIFF creation options?

smathermather, 2017-03-31:
I always use average, as it is fast and good enough for pyramids. I would ignore -b(yes it will use all bands if not declared). I see no great advantage to -ro and levels I'd keep to 2 4 8 16. Any more, and depending on how small the dataset is, GDAL can hang.

I don't have a strong opinion on exposing these parameters. On one hand it's nice. On the other hand, why add complexity, when those that are really interested in these features can run gdaladdo themselves.

dakotabenjamin, 2017-03-31:
So trying on a BIG dataset, I run into issues, even when trying to set BIGTIFF. Not really sure what's wrong. 
```
H:\Junk\alces_flight_ewinters>gdaladdo -r average --config BIGTIFF_OVERVIEW YES
odm_orthophoto.tif 2 4 8 16
0ERROR 1: TIFFWriteDirectoryTagData:Maximum TIFF file size exceeded. Use BIGTIFF
=YES creation option.
ERROR 1: odm_orthophoto.tif:Failed to allocate memory for to read TIFF directory
 (0 elements of 12 bytes each)
ERROR 1: TIFFReadDirectory:Failed to read directory at offset 4294965964
ERROR 1: odm_orthophoto.tif:Failed to allocate memory for to read TIFF directory
 (0 elements of 12 bytes each)
ERROR 1: TIFFReadDirectory:Failed to read directory at offset 4294965964
Overview building failed.
```

dakotabenjamin, 2017-03-31:
I think it was just the file may have had some issues. I'm trying with other stuff.

smathermather, 2017-03-31:
This is where setting the if safer option may be a good idea, as gdaladdo will make the tiff even bigger. Or we play it safe and always use the -ro flag. Hmm, compression on overviews is a good idea too. This might be set with -oo. Setting even to JPEG is safe, since we aren't modifying the original data.

dakotabenjamin, 2017-03-31:
The spec page doesn't really go over -oo so I don't know how to apply it. 

To me it looks like compressions options is set with --config when specifically working with GeoTIFFs.

smathermather, 2017-03-31:
Flags would be anything under creation options here I think: http://www.gdal.org/frmt_gtiff.html , but --confie should work too.

dakotabenjamin, 2017-03-31:
It looks like -ro is necessary

dakotabenjamin, 2017-03-31:
@pierotofy I'd love your input since this is useful to WebODM. Right now I'm going to put the following in:
```
gdaladdo -ro -r average --config BIGTIFF_OVERVIEW IF_SAFER --config COMPRESS_OVERVIEW JPEG {orthophoto} 2 4 8 16 > {log}
```

pierotofy, 2017-03-31:
It looks good to me, and I second @smathermather opinion that we should keep the number of flags under control, exposing only the really important ones.

pierotofy, 2017-03-31:
average sampling for overviews is fine, I don't think we need to expose the option (or at least until somebody requests it).

dakotabenjamin, 2017-03-31:
Right now I'm not exposing any.

dakotabenjamin, 2017-03-31:
pushed to #525

closed 2017-04-04 by dakotabenjamin