OpenDroneMap/ODM #528: Newbie question - Errors after "DOCKER BUILD -T ODM_IMAGE ." and various Vtk Library reference errors

ProjectNeurogenesis, 2017-04-01:
Hi. I'm trying to build ODM from Dockeralready followed the preliminary steps re: "Building ODM with Docker"  and encountered several errors after "$docker build -t odm_image ."

The following types of errors were displayed at the Docker terminal
[image]

Also it was indicated that Vtk Library / module was not found. How do I install them and where. I tried searching online but most of the links had voluminous content of texts I can't 'grasp' well since I am new in the Git / Docker / development world :P =D. Can anyone help me re: what I should do to resolve these issues?
[image]

Thank  you in advance. Immediate responses much much appreciated.

pierotofy, 2017-04-01:
Those look like warning messages. I wouldn't worry about them. Did the build process complete?

ProjectNeurogenesis, 2017-04-02:
It wasn't stated that the build process was completed so i typed in "$sudo docker images ls -a" and the images displayed were incomplete compared to the tutorial video shown below the web odm repo at this site...I also tried installing it with an Ubuntu 16.04 VM on my Windows 10 laptop but the following error now displays...
[image]

pierotofy, 2017-04-02:
You don't need to run Ubuntu on Windows 10 via a VM. I'd recommend deleting any files / folders / VMs you have setup so far, then in Windows 10, from the Docker Quickstart Terminal, simply type:

```
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input
cd WebODM
./webodm.sh start
```

This should take care of everything for you. 👍

ProjectNeurogenesis, 2017-04-02:
Actually that was the first thing I did before installing any VM on windows 10, Docker first :). However the docker terminal displayed those missing Vtk libraries ad stuff :). Im currently downloading Vtk and Cmake and will try to reconduct the building process once I've tried installing those two.

Hoepefully they'll be gone. Than

dakotabenjamin, 2017-04-03:
So now are you trying to run ODM natively or with WebODM?

ProjectNeurogenesis, 2017-04-05:
I installed WebODM natively. Currently trying to process a roadway mapping flight of 123 imgs. No problems so far thanks :). 
Now my major inquiry is whether there a script / method to generate a DEM together with the Orthomosaic? Is there a 'Web GUI" (sry I don't know the term) with functions which lets a user input a .txt / .csv file containing surveyed Ground Ctrl Points s and tick those corresponding gcp markers in the previously generated orthophoto / set of photos?

If there's a way, can you enumerate step by step (newbie here) on how I can incorporate those scripts / codes into the Default processing node?

Thank you in advance.

PS: This is for an undergraduate research btw. Originally planned to use Cmpmvs but then discovered that it's not really 'open source' in the technical definition :) Hence, I gave webodm a try.

pierotofy, 2017-04-05:
The DEM is one of the next features that are going to be rolled out soon. @dakotabenjamin is working on the ODM side of it, and as soon as it's ready, I will work to include it in WebODM.

We have a contrib script https://github.com/OpenDroneMap/OpenDroneMap/tree/master/contrib/grass that can generate a DEM in the meanwhile, but it's not supported by WebODM.

Ground Control Points are another feature that we are working on, see https://github.com/OpenDroneMap/WebODM/issues/37. We plan to have a UI that will allow users to tick GCP markers like you described.

You can manually feed a ground control point file using ODM or node-opendronemap, but there's no support for it in WebODM at the moment.

ProjectNeurogenesis, 2017-04-06:
Thank you. Will understand the GRASS script and try implementing it in a VSfM->Dense cloud->DEM and Meshes workflow.
Btw, are there any significant differences from DEMs derived from Meshes and DEMs generated from an ASCII file or an .LAS file?

dakotabenjamin, 2017-04-06:
The DEM will be derived from the LAS file generated in ODM. I've got the code up here: https://github.com/dakotabenjamin/OpenDroneMap/tree/pdal-dem but I ran into an issue with the installation process that I must address first before I can merge into master.

ProjectNeurogenesis, 2017-04-11:
Hi. This isn't related to the question above but I'll just ask if the
algorithm of Open Drone Map or Web Open Drone map for radial distortion or
calibration is better than / similar to the one used by Visual SfM?
(Question arose as a result of doming errors)

Thanks!

smathermather, 2017-04-11:
Barrel distortion is still an issue, but may go away once we pull in the latest OpenSfM, see: https://github.com/OpenDroneMap/OpenDroneMap/issues/538#issuecomment-293252306

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/newbie-question-errors-after-docker-build-t-odm-image-and-various-vtk-library-reference-errors/374

closed 2018-03-28 by dakotabenjamin