OpenDroneMap/ODM #532: Help with smoothing model

mribbons, 2017-04-04:
Is there a way to smooth the model which is generated from the point cloud?

The point cloud looks good, but there is definitely some noise. Is there a way to reduce the noise, or possibly make an assumption that surfaces are going to be flat or regular curves, as opposed to random/fractal shapes? (My problem domain isn't concerned with vegetation).

What I'm hoping for is a building with straight edges rather than blobs.

[image]

[image]

dakotabenjamin, 2017-04-04:
What's the vertex count on the mesh? You could decimate the mesh, which would sacrifice some accuracy for less noise. 

Here's the parameters that can affect the mesh: 
```
  --mesh-size <positive integer>
                        The maximum vertex count of the output mesh. Set to 0 for no decimation. Default:
                        100000
  --mesh-octree-depth <positive integer>
                        Oct-tree depth used in the mesh reconstruction,
                        increase to get more vertices, recommended values are
                        8-12. Default: 9
  --mesh-samples <float >= 1.0>
                        Number of points per octree node, recommended and
                        default value: 1.0
  --mesh-solver-divide <positive integer>
                        Oct-tree depth at which the Laplacian equation is
                        solved in the surface reconstruction step. Increasing
                        this value increases computation times slightly but
                        helps reduce memory usage. Default: 9
```

The mesh will still be "blob" shaped, because of the poisson surface reconstruction. Things I think that would help that are currently outside the scope of ODM functionality are to reduce the noise of the point cloud and use an algorithm that better reconstructs flat surfaces. For the first, you could do this manually and reinsert into ODM using `--rerun-from odm_meshing.` For the second, there is still lots of research to be done.

pierotofy, 2017-04-04:
I think the mesh reconstruction could be improved, in particular there should be a outliers removal step on the point set before we do meshing. Decimating the mesh will reduce the number of triangles, but will not necessarily remove outliers.

@mribbons if you need better control over the reconstruction, download the point cloud assets (ply, las) and use Meshlab to finely tune the parameters. You'll be looking at denoising the point cloud (also check out PDAL in this regard https://www.pdal.io/stages/filters.outlier.html) and running a poisson reconstruction.

mribbons, 2017-04-05:
Thank you both for your suggestions.

Unfortunately I am unable to try them as the feasibility study I was working on has concluded.

kikislater, 2017-04-06:
Is it possible to re open ? Sound like very interesting
Does anyone know the PCL process listed here : https://github.com/PDAL/PDAL/issues/1553#issuecomment-292036719

dakotabenjamin, 2017-04-06:
THat's just the export function for mesh files. I don't know all the pcl meshing algos; we use Poisson Surface Reconstruction: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_meshing/src/OdmMeshing.cpp#L273

kikislater, 2017-06-30:
New non PCL Poisson reconstruction in PDAL  :  
https://github.com/PDAL/PDAL/pull/1618

dakotabenjamin, 2017-06-30:
With this we may be able to completely remove pcl, which is a massive dependency. I'll check it out!

PDAL/PDAL#1618<https://github.com/PDAL/PDAL/pull/1618>

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/help-with-smoothing-model/402

closed 2018-03-28 by dakotabenjamin