OpenDroneMap/ODM #534: Full Package audit

dakotabenjamin, 2017-04-04:
We've been updating the modules continuously but haven't been keeping up too well with the dependencies list. A full package audit will help reduce the size of the ODM, including the docker image. See #461

KommandorKeen, 2017-07-26:
What's involved in the audit and do you need volunteers?

dakotabenjamin, 2017-07-26:
It involves going through each dependency in configure.sh and the Dockerfile and finding out where it is or was used in ODM. If that package or w/e is irrelevant, then remove it. It's a fairly straight-forward task but requires some in-depth knowledge of the whole process, or a willingness to be thorough. If you're up to the task, by all means we need all the help we can get.

closed 2020-04-14 by smathermather