OpenDroneMap/ODM #535: Problem with Mavik data set

Ekci, 2017-04-04:
I have a set of data from flat terrain which I processed in ODM and Pix4D. In both cases the altitude heat map looks like this:
[image]
[image]
when the terrain should be more or less flat, I've checked datasets from different places and they all generate curved model. This made me assume that the camera's EXIF tags are wrong so I played with them: setting focal length to 8 in Pix4D helped to generate flat terrain but using `--force-focal 8.0` in ODM changes nothing.  I've also tried to add `--force-ccd 6.17` but without luck.

Has any one had a problem like this? What numbers should I set?

smathermather, 2017-04-04:
There are two different common reasons for this happening, and none are related to GPS (but stay tuned, as GPS _may help_ in the future with one of the cases).

The first possibility is that the optical correction parameters aren't addressing adequately all the lens correction that needs done for the dataset. This is common with DJI cameras, GoPros, and other wide angle lenses that require more parameters for correction.  The work around at the moment is to pre-calibrate your images. @dakotabenjamin has a script for this, and it works really well.

The second possibility is that your flight path was very flat, and your terrain is and surface are very flat (not a lot of trees, buildings, etc. although it looks like there is some of each), and so there can be systematic bias in the reconstruction of the scene. This often presents itself as a bowl like shape, as small errors in any set of matches get compounded (added) across the scene. There are operational ways to fix this and software ways to fix it.

The operational trick is to flight plan for image collection at two different heights. This ensures that the reconstruction software can infer correct Z values, as there's a higher signal to noise ratio along your Z axis.

The software fix is to use the GPS positions to help initialize the model, which may help in determining the overall global model, and compensate for smaller matching errors. We are working toward this end. Is this a dataset you can share (publically or privately) as a good test case for bias issues in reconstruction?

Ekci, 2017-04-04:
Great answer, thank you. I will try adding additional photos from higher altitude and update here.

I can privately share the dataset: please tell me how can I send you a link :)

Ekci, 2017-04-04:
Here's map generated using one dataset from 40 m and one from 80 m. Second dataset covered only top of the map but it still looks much better (if not perfect). Thank you, good job! (still waiting for a way to send you my pictures)

kikislater, 2017-04-05:
What is your setup ?
Gimbal camera is very bad for photogrammetry but usually you have a dome effect and not a reverse dome ...

smathermather, 2017-04-06:
Hi @Ekci, please send a link to svm at clevelandmetroparks dot com. This does look much better.

dakotabenjamin, 2017-04-06:
Also depending on your camera, you can try to use the [Lensfun](http://lensfun.sourceforge.net/) library which will remove distortion.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/problem-with-mavik-data-set/397

closed 2018-03-28 by dakotabenjamin