OpenDroneMap/ODM #539: ODM install ERROR (Docker): "c++: internal compiler error: Killed (program cc1plus)"

ProjectNeurogenesis, 2017-04-06:
Ran Docker and followed all command inputs outlined in https://github.com/OpenDroneMap/OpenDroneMap#build-and-run-using-docker. 
Everything was built successfully except during this last part in which the ff were displayed (am I the only one encountering these kinds of errors in recent odm installations? T_T)
[image]

Attached below is a .txt file of the LOG from the installation. The above mentioned error can be found at Line: 6054

Thank you.#

dakotabenjamin, 2017-04-06:
I don't think the txt file attached. 

I've never seen that error before... But whenever I see a "Killed" error I think that it is a memory issue. Have you allocated enough memory to Docker? @pierotofy maybe we should add those instructions to the readme...

pierotofy, 2017-04-06:
Yes, the build failed because the machine ran out of memory. Make sure your docker machine has enough memory allocated.

And yes we should add a note in the readme.

stiward, 2017-08-02:
I have the same problem, how can I solve it?
How can I allocate more virtual memory to Docker

Tnks you

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/odm-install-error-docker-c-internal-compiler-error-killed-program-cc1plus/398

closed 2018-03-28 by dakotabenjamin