OpenDroneMap/ODM #545: Exception: Child returned 1 while --rerun-from opensfm

specyfick, 2017-04-15:
Hi,
I have suffered today that problem.
I've executed first command:
_python run.py --project-path /home/specyfick/ODMProjects/ --orthophoto-resolution 20 --rerun-from odm_orthophoto --force-ccd 6.17 ProjektLakiShort_

All process went without any problem. Mesh was generated and orthophoto.
Unfortunately images rectification was not good and images have a small distortion remaining.
I wanted to correct K1 and K2 parameters in camera_models.json with camera_models_override.json file in /project/name/opensfm/ means in project folder.
I'm using Hero3 Black camera, and distortions parameters estimated by a ODM are not meet my requirements. I have got my own calibration results and i want to put them to a file.

As I understood, i can correct file above and rerun process again. I've rerun it with command:
_python run.py --project-path /home/specyfick/ODMProjects/ --orthophoto-resolution 20 --rerun-from  opensfm --force-ccd 6.17 ProjektLakiShort_

So from from Opensfm.
No files corrected. Just rerun for test. And recieved error:
 
_[DEBUG]   running /home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon /home/specyfick/ODMProjects/ProjektLakiShort/opensfm/reconstruction.nvm /home/specyfick/ODMProjects/ProjektLakiShort/odm_meshing/odm_mesh.ply /home/specyfick/ODMProjects/ProjektLakiShort/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Apr 13 2017, 21:15:45)
Load and prepare mesh: 
PLY Loader: comment PCL generated
Reading PLY: 493703 verts... 988845 faces... done.
Warning: Zero-length normals detected: 988783 face normals, 493568 vertex normals
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 11
NVM: Number of features: 0
        Loading 100%... done. (Took 4.2s)                                       
Building adjacency graph: 
        Adding edges 100%... done. (Took 2.992s)                                
	1483159 total edges.
View selection:
	Building BVH from 988845 faces... done. (Took: 284 ms)
        Calculating face qualities 100%... done. (Took 2.623s)                  
        Postprocessing face infos 100%... done. (Took 0.073s)                   
	Maximum quality of a face within an image: 0
	Clamping qualities to 0 within normalization.
	Writing data cost file... done.
	988845 faces have not been seen by a view.
	Took: 15.171s
Generating texture patches:
	Running... done. (Took 2.157s)
	0 texture patches.
Running global seam leveling:
	Create matrices for optimization...  done.
	Lhs dimensionality: 0 x 0
	Calculating adjustments:
Received signal SIGSEGV (segmentation fault)Received signal SIGSEGV (segmentation fault)
Obtained 6 stack frames: 0x4de294 0x7fad0b8f74b0 0x47dce4 0x7fad0c0cd43e 0x7fad0bc926ba 0x7fad0b9c882d_

Obtained 8 stack frames: 0x4de294 0x7fad0b8f74b0 0x47dce4 0x7fad0c0c9cbf 0x47b6b3 0x42846e 0x7fad0b8e2830 0x42a9a9
Received signal SIGSEGV (segmentation fault)
Obtained 6 stack frames: 0x4de294 0x7fad0b8f74b0 0x47dce4 0x7fad0c0cd43e 0x7fad0bc926ba 0x7fad0b9c882d
/home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon/home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon(_ZN4util6system17print_stack_traceEv+0x34)[0x4de294]
(/lib/x86_64-linux-gnu/libc.so.6(+0x354b0)[0x7fad0b8f74b0]
_ZN4util6system17print_stack_traceEv/home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon[0x47dce4]
/usr/lib/x86_64-linux-gnu/libgomp.so.1(GOMP_parallel+0x3f)[0x7fad0c0c9cbf]
+0x/home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon(_ZN3tex20global_seam_levelingERK8UniGraphSt10shared_ptrIKN3mve12TriangleMeshEERKNS4_8MeshInfoERKSt6vectorISB_INS_20VertexProjectionInfoESaISC_EESaISE_EEPSB_IS3_I12TexturePatchESaISK_EE+0x1993)[0x47b6b3]
34/home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon(main+0xa9e)[0x42846e]
)/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7fad0b8e2830]
/home/specyfick/OpenDroneMap/SuperBuild/install/bin/texrecon(_start+0x29)[0x42a9a9]
Segmentation fault
[0x4de294Traceback (most recent call last):
  File "run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/specyfick/OpenDroneMap/scripts/mvstex.py", line 109, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/specyfick/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

Any help will be appreciated.

dakotabenjamin, 2017-04-17:
Hi,

Can you try deleting the odm_texturing folder and then running the same command? Also please post what version of ODM you are using. Thank you for the detailed post.

yjmenezes, 2017-05-05:
@specyfick 

I am running:  ODM ver 0.3  from git 2017-05-04 under Ubuntu 16.04

for me,  opensfm crash and report:

[DEBUG]   running PYTHONPATH=/mnt/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /mnt/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm mesh /mnt/odm_terra_nova/opensfm
Traceback (most recent call last):
  File "/mnt/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/mnt/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/mnt/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/mnt/odm_terra_nova/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "/mnt/OpenDroneMap/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/mnt/OpenDroneMap/scripts/opensfm.py", line 112, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/mnt/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

please, if possible, could you or someone test with my dataset ?
I have a good dataset, available here:

https://github.com/yjmenezes/odm_tnova

--force-focal 152.274 --force-ccd 206.322   --use-exif

thank you

dakotabenjamin, 2017-05-08:
@specyfick any luck?

dakotabenjamin, 2017-05-08:
@yjmenezes ill test that now

specyfick, 2017-05-08:
@dakotabenjamin at the moment no time to test, but will do asap

yjmenezes, 2017-05-08:
@dakotabenjamin,   thank you.

As i have contours from a Photogrammetric Workstation, i wanna test morphological filter.

I am still trying to figure out the opensfm crash reason.

dakotabenjamin, 2017-05-08:
What's likely happening is an inability to match the images either due to lack of overlap or some other reason. Maybe also related to the camera. If there is large distortion it could also fail to create a reconstruction.

yjmenezes, 2017-05-08:
@dakotabenjamin

I have cropped from 9728 to 8600 but  i think side overlap  is still greater than 20%.
The lens distortion are very small.

**Does it make sense ?
camera_models.json   file:
{
    "v2 lmk  8600 8600 perspective 0": {
        "focal_prior": 0.85, 
        "width": 8600, 
        "k1": 0.0, 
        "k2": 0.0, 
        "k1_prior": 0.0, 
        "k2_prior": 0.0, 
        "projection_type": "perspective", 
        "focal": 0.85, 
        "height": 8600
    }
}

**And this:  "focal_ratio": 0,  ?

01_001.jpg EXIF
{
    "width": 8600, 
    "camera": "v2 lmk  8600 8600 perspective 0", 
    "projection_type": "perspective", 
    "orientation": 1, 
    "focal_ratio": 0, 
    "make": "LMK", 
    "gps": {
        "latitude": -8.52380510000434, 
        "altitude": 2785.308, 
        "longitude": -40.04354469999571
    }, 
    "model": "LMK", 
    "capture_time": 0.0, 
    "height": 8600
}

yjmenezes, 2017-05-08:
@dakotabenjamin

thx, 

I have cropped from 9728 to 8600 but i think side overlap is still greater than 20%.
The lens distortion are very very small,   less than 20 microns.

**Does it make sense ?
camera_models.json file:
"v2 lmk 8600 8600 perspective 0": {
"focal_prior": 0.85,

**And this: "focal_ratio": 0, ?

01_001.jpg EXIF
"focal_ratio": 0,

dakotabenjamin, 2017-05-08:
We typically run sidelap of 60% minimum for photogrammetry, but your mileage may vary. The crop may actually have an effect. Have you tried no cropping but resizing instead?

yjmenezes, 2017-05-08:
@dakotabenjamin,  thx very much.

I will try resize, 

I have tried to keep the original 24 microns image quality, avoiding image resampling.
I guess the flight plan was trivial 60% overlap vs 30 % sidelap used for analog camera.

yjmenezes, 2017-05-12:
@dakotabenjamin 

i replace the old 8600x8600 by  a 9500x9500 images without exiftags. 
Now it works, thx you.

I realised:

GPS exiftags are the OpenSfm crash reason.

1- I run ODM 0.3.1 with defaults 2400x2400 resize. 

2- gcp points with 2 images can not be used for georeferencing, at least for opensfm model.
please take a look on footprint.

3- i have ploted some footprints of opensvm vs pmvs

the absence of GPS Exif do not affect the process but can increase photo alignment  runtime , right ?

Could camera projection centers be loaded thru files like gcp_list.txt ?

thx

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/exception-child-returned-1-while-rerun-from-opensfm/400

closed 2018-03-28 by dakotabenjamin