OpenDroneMap/ODM #546: Playing with resize-to and  orthophoto-resolution

clarkerz, 2017-04-15:
Hi,

Firstly **THANK YOU** so much for the effort and hard work you have gone to in developing ODM and WebODM!! It's greatly appreciated!! I managed to get both working (latest code form here)  on an old AMD i3 dual booted laptop with Ubuntu 16.04, 4Gb RAM and 4 cores. It's not the most powerful but I've added a 64Gb swapfile on a USB3 which it rarely uses when processing the images. I wanted to try and improve the output of the orthophoto as it became a bit "pixely" when zooming into to 25M when imported to Google Earth. I read on here that reducing the resize and increasing the orthophoto-resolution can improve the output. I probably am being a bit picky as the output with the default 2400 resize is really good in any case.

I thought I'd share them on here (I hope that's OK) as I don't have an issue but more of a question really.

I used my DJI P4 rather than the Inspire as it's quicker to setup and used a forced-ccd of 20 on all 3 tests ( I read that somewhere that was correct but been able to revalidate again). For a comparison I used three different settings and zoomed into 25M on Google Earth with each imported tif and took a screenshot  so I am hoping that is a valid test?

The first was all settings default:
[image]

This is the output  second with resize-to 3950 and  orthophoto-resolution of 30:
[image]

This is the output  second with resize-to 3999 and  orthophoto-resolution of 60 which I think is awesome!!:
[image]

Hopefully I am doing this right but wasn't sure what the max and min parameters are the for the orthophoto-resolution but 60 appear to work really well. I couldn't find a don't resize option but when I set to 4000 it complained.

Is there anyway to improve the 3D mapping. To reduce the number of holes etc and also  the side of buildings etc and does it help to photo the buildings at a lower level as the guys at Drone Deploy suggest?   Hope it's OK to post this here.. apologies if it's not. 

Thanks again!

Cheers

Andy

pierotofy, 2017-04-15:
Hey thanks for sharing the images Andy! They look pretty good!

Changing the orthophoto-resolution parameter is correct, but you don't need to change the resize-to parameter.

To get the side of the buildings you need to shoot photos from the side (example: set your target, then use the orbit functionality to take pictures from the side). Drone Deploy has a few articles on how to get that done.

We are currently working to improve the orthophoto results here #521. If you want a sharper textured model, increase the `mesh-octree-depth` parameter (set it to `12` or slightly higher).

clarkerz, 2017-04-16:
Thanks for the information Pierotofy, greatly appreciated. That sounds great about improving the results and if you need any testing let me know! :)

Is there anywhere that explains each parameter and the max and min for each and what it does? i.e. what is the max for orthophoto-resolution etc? I'll try the mesh-octree-depth as well. Thanks again.

pierotofy, 2017-04-16:
If you run:

`python run.py --help`

Reading the descriptions is the best documentation we have at the moment. You can also look at the source here: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/opendm/config.py

I think we're looking to improve the documentation of the parameters as part of #516

LucasMM86, 2017-04-16:
orthophoto-resolution is how many pixels per meter you want on the orthophoto.

clarkerz, 2017-04-17:
Thanks both. I've used the Web DroneDeploy service just for some comparisons and they use the size of the area per pixel which confused me a bit i.e.  0.4in/px or 2in/px... as opposed to the number of pixels per area... I was just being a bit **thick** now I've worked it out. :) I am loving OpenDronemap!!! :)

Clarkerz

clarkerz, 2017-04-18:
Just testing some more and it looks like the if I make the resize 3999 then the ouput file is much better. Also when I change the mesh-octree-depth to 12 I get a Segamentation Fault and then WebODM core dumps and if I try 11 then it bombs out and hangs. I'll leave at 9 as that works well.

orthophoto-resolution of 70, No forced-ccd and no resize
[image]

orthophoto-resolution of 70,  forced-ccd 20 and no resize
[image]

orthophoto-resolution of 70,  forced-ccd 20 and resize 3999 . I am well impressed with this as this is VERY zoomed in!! :)
[image]

closed 2017-04-20 by clarkerz

dakotabenjamin, 2017-04-20:
At some point you will reach the limit of the textured mesh's resolution.

clarkerz, 2017-04-21:
Thanks. I reckon it's about 60..70 doesn't improve. Thanks Andy. It's still good though! :)

BuffMyRadius, 2020-08-10:
Something that I would like to add for anyone searching for answers on how to up output resolution is that you can disable the input image resize with `resize-to -1`.

shelhart, 2022-08-26:
Just wanted to say that this articlle (5 years later) has been EXTREMELY helpfull.
Thanks @clarkerz :)