OpenDroneMap/ODM #550: Question: Re log on Web ODM

clarkerz, 2017-04-23:
Hi,

is it possible to change the GMT time  so that the rolling log in the web browser shows the local time (in my case BST)?

Thanks

Andy

dakotabenjamin, 2017-04-24:
Are you using WebODM? That's a good question for @pierotofy

pierotofy, 2017-04-24:
I assume that you mean this log:

[image]

In that case, at the moment we have no way to set a GMT time. The log is simply the output taken from ODM. This is probably something that needs to be implemented in node-odm.

@clarkerz could you open an issue here https://github.com/OpenDroneMap/node-OpenDroneMap as a enhancement request?

dakotabenjamin, 2017-04-25:
Closing as it applied to https://github.com/OpenDroneMap/node-OpenDroneMap

closed 2017-04-25 by dakotabenjamin

clarkerz, 2017-04-26:
Apologies as I have been away for a few days. Have just raised in node-OpenDroneMap as request.

Thanks

Andy