OpenDroneMap/ODM #556: OpenCV cmake error during installation

macsee, 2017-05-01:
Hello!. I was trying to install ODM on an Ubuntu 17.04 machine without success. I got this error related to OpenCV and I don't know how to solve it. Any clues?
Thanks!

Cheers.

CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

dakotabenjamin, 2017-05-01:
Hey thanks for the info. There are multiple issues right now with 17.04 and I'm working to fix them by the next update.

macsee, 2017-05-01:
Oh! I see. In the meantime Can you tell me wich is the latest or the best Ubuntu version for working with ODM please? Thanks!

dakotabenjamin, 2017-05-01:
16.04 is what I use and recommend. Alternatively you should be able to get it running in Docker from any Ubuntu version. See: https://github.com/OpenDroneMap/OpenDroneMap/wiki/Docker

macsee, 2017-05-01:
Ok thanks! I'll try that too.
Cheers!

jirgg, 2017-05-07:
I'm having the same issue trying to install in Ubuntu 16.04.2 on a VirtualBox installation. I tried the Docker route and received the same messages. Also getting a problem with the PCL config.

CMake Warning at modules/odm_georef/CMakeLists.txt:16 (find_package):
  Could not find a package configuration file provided by "PCL" (requested
  version 1.7) with any of the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

dakotabenjamin, 2017-05-08:
Hm, I just tried a 16.04.2 fresh install with no problems. What version of ODM are you trying to install?

jirgg, 2017-05-08:
It must be v0.3, I was trying to follow the latest instructions on this site.

I got the Docker build up and running at least,

irishhawk23, 2017-05-25:
Hello I am having the same issue as above. 

[ 56%] Completed 'ceres'
[ 56%] Built target ceres
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Compiling build
CMake Warning at modules/odm_georef/CMakeLists.txt:16 (find_package):
  Could not find a package configuration file provided by "PCL" (requested
  version 1.7) with any of the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

I am using 16.04 and would prefer not to have to use docker for this. Is this something that is looking to be fixed? 

I appreciate the help.

dakotabenjamin, 2017-06-07:
@irishhawk23 with compilations that happen over multiple processes sometimes the error happens further up the stdout. Can you search the output for where the make error happened to see which part failed?

dakotabenjamin, 2017-06-07:
There should be another error somewhere higher in the stdout of the compilation process. Can you check?

PeterSprague, 2017-10-13:
I'm getting the same ending error when building inside a nvidia/cuda:9.0-devel-ubuntu16.04 container.  Not erroring prior in the build.

CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/OpenDroneMap/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/OpenDroneMap/build/CMakeFiles/CMakeError.log".

talltom, 2018-01-04:
@dakotabenjamin To help support this issue, I've setup Travis builds of ODM on the `travis` branch (created from master).

The OpenCVConfig.make configuration file issue occurs in the Travis Trusty Environment (Ubuntu 14.04) - see this build: https://travis-ci.org/OpenDroneMap/OpenDroneMap/builds/325217072

Next up I plan to create automated builds using Ubuntu 16.04 (using Travis' docker images).

talltom, 2018-01-05:
Hi @dakotabenjamin, I can confirm that the OpenCV issue reported above also occurs in Ubuntu 16.04.03. I've setup Travis to build from fresh Ubuntu images, and you can see the output here: https://travis-ci.org/OpenDroneMap/OpenDroneMap/builds/325598188 (note Travis is reporting success as the build process in the docker container isn't returning an error code).

The good news is that your pull request (https://github.com/OpenDroneMap/OpenDroneMap/pull/677) seems to fix the issue, and successfully builds in this environment.

gowthamaraj-creator, 2019-12-11:
cmake -DCMAKE_BUILD_TYPE=Release ..
CMake Deprecation Warning at CMakeLists.txt:31 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:38 (cmake_policy):
  The OLD behavior for policy CMP0022 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Error at CMakeLists.txt:88 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" (requested
  version 2.4.5) with any of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/karthi/openbr/build/CMakeFiles/CMakeOutput.log".
   help!!!

smathermather, 2020-04-14:
I think this has been long a non issue. Closing.

closed 2020-04-14 by smathermather