OpenDroneMap/ODM #557: ODM to display the camera angles to associate images with 3D Mesh

shaunktw, 2017-05-01:
Pix4D has a Ray Cloud feature to cut the thumbnails of the cameras at the location where the point is visible in the original images. An example: https://support.pix4d.com/hc/en-us/articles/202557999-Menu-View-rayCloud-3D-View#gsc.tab=0

dakotabenjamin, 2017-05-01:
So I think we can extract the camera position and angle from opensfm/reconstruction.json: 
```
  "shots": {
            "DSC00298.JPG": {
                "orientation": 1, 
                "camera": "v2 sony dsc-wx220 1800 2400 perspective 0", 
                "capture_time": 1467198160.0, 
                "gps_dop": 15.0, 
                "rotation": [
                    2.4804502253475951, 
                    -1.5131942004329673, 
                    -0.15625880101596284
                ], 
                "translation": [
                    -19.708380746486213, 
                    -46.029045816851102, 
                    14.960096193751962
                ], 
                "gps_position": [
                    -31.69669311275203, 
                    -40.237064567700145, 
                    1.9997939048334956
                ]
            }, 
```

It's a matter of exporting that to a usable format and then joining that with info from the texturing data. This could be solved in part in ODM and in part in WebODM with @pierotofy .

pierotofy, 2017-05-01:
`reconstruction.json` could be probably imported in the potree viewer in WebODM, so visualization of the cameras shouldn't be too difficult to add. Tying in the rays would be a little more involved.

@shaunktw perhaps open an issue in the WebODM repository, as it's probably the right place to ask for an enhancement request.

dakotabenjamin, 2017-05-01:
I think tying the camera to a face will be the most challenging. I'm going to ping the mvs-texturing guys to see if we can extract that info.

fredlllll, 2017-05-01:
what should be used for the camera position? gps_position or translation?

dakotabenjamin, 2017-05-01:
Good question! I *think* you have to use both, you must translate the position to match the reconstruction. But I've not tried to do that.

fredlllll, 2017-05-01:
seems strange that they have to translated again. what are the gps coordinates referenced to in the first place then? i will try both tomorrow and then report back with what is right

dakotabenjamin, 2017-05-02:
Probably because the sparse point cloud is unreferenced.

fredlllll, 2017-05-02:
ok this is using only the gps_position http://i.imgur.com/r5SS5nP.png and this is using only the translation: http://i.imgur.com/3KluTBn.png

i have no idea what the hell translation does. it seems that gps_position works fine though

fredlllll, 2017-05-03:
currently struggling with the rotation. i have a feeling the rotation data is completely unrealiable. looking at this here it looks like the cameras are just pointing in random directions, while they were most probably all pointing down. blue lines are vec3(0,1,0) rotated using the rotation of reconstruction.json http://i.imgur.com/A1y32yU.png

pierotofy, 2017-05-03:
OpenSfM already displays the camera positions with their built-in viewer, so perhaps look how they do it in their source: https://github.com/mapillary/OpenSfM/blob/master/viewer/reconstruction.html

fredlllll, 2017-05-04:
that helped regarding the camera rotation. seems like translation and rotation together give the cameras rotation and position, but they have to be mixed together (see the code you linked). i initially thought that rotation would just be an euler, but it seems to be a rotation around an axis that is used on the translation. i wonder why they didnt just include the optical center and the rotation as a quaternion. would be much less confusing

Next issue:
GPS Altitude

it seems the gps_positions z value is almost the same for every shot http://i.imgur.com/9Xzwi8y.png but the actual gps altitude from the exif data varies greatly. http://i.imgur.com/eWh3Yph.png

is this intended behaviour or a bug?

LucasMM86, 2017-05-04:
how did you take the images? I belive some UAV use the barometric pressure to maintain altitude, because the GPS position has a bigger error. So maybe the actual camera positions are almost the same for every shot, even if the exif GPS data varies so much.

fredlllll, 2017-05-04:
if it is called gps_position it should use the gps data right? i dont know anything about the drone that took the photos

dakotabenjamin, 2017-05-04:
For the Bellus dataset, the images were taken on a Sony WX-220 on a SenseFly eBee drone. The images are referenced by the flight planning software using drone logs. IIRC the eBee does use barometric pressure to monitor altitude.

gogikomal, 2017-09-28:
Hey, 
I'm using OpenDroneMap on Linux 16.04, it works perfectly fine. The output of data-set is viewed in mesh-lab, i see that the point cloud and texturing image is not very satisfying. I tired changing the parameters in Opendronemap such as 
--min_features: 50000
--mesh_size: 4000000
--resize: 4000
--mesh_octree_depth: 11
--mesh_solver_divide: 11
Still there are no effect on the texturing image.
is there any other parameter to get a prefect 3D model ?
or is there any other software to view 3D Model and also supports OpenDroneMap ?
How can i get control over point cloud?

garlac, 2018-11-27:
I am trying something similar - display cameras over the point cloud and on clicking them image should display over point cloud.

I used the rotation matrix from bundle_r000.out and mapping it with img_list.txt

However, though the cameras look like aligned properly, the images are not displayed properly.There seems to be axes orientation issue..

pierotofy, 2020-09-11:
ODM now generates a `shots.geojson` in `odm_report` that can be used to display the camera angles/positions in a 3D viewer (WebODM already implements that, and displays thumbnails too..).

If something was missed and still needs to be implemented, please re-open? :pray:

closed 2020-09-11 by pierotofy

smathermather, 2020-09-12:
I think this is broadly complete. Pix4D does some cool stuff with tracing rays back to their origin etc., but I get the sense it's mostly just for oohs and ahhs.