OpenDroneMap/ODM #558: Compilation fails on Ubuntu 14.02

johnfox, 2017-05-02:
Hi,

I'm trying to compile latest version 0.3 RC and the compilation fails on:

`
In file included from /usr/local/include/boost/atomic.hpp:12:0,
                 from /usr/local/include/boost/thread/pthread/once_atomic.hpp:20,
                 from /usr/local/include/boost/thread/once.hpp:20,
                 from /usr/local/include/boost/thread.hpp:17,
                 from /usr/include/pcl-1.7/pcl/io/boost.h:50,
                 from /usr/include/pcl-1.7/pcl/io/file_io.h:43,
                 from /usr/include/pcl-1.7/pcl/io/obj_io.h:42,
                 from /home/fox/open_drone_map/modules/odm_orthophoto/src/OdmOrthoPhoto.hpp:9,
                 from /home/fox/open_drone_map/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:7:
/usr/local/include/boost/atomic/atomic.hpp:202:16: error: ‘uintptr_t’ was not declared in this scope
 typedef atomic<uintptr_t> atomic_uintptr_t;
`

I'm guessing this is boost library version problem - I'm sure I have latest, but would appreciate any help with compilation or any workaround to include right boost version.

Should I try on different Ubuntu version? I was following recommendation on the project home page saying Ununtu 14 is the right one :)

Cheers and thank you for great job.

johnfox, 2017-05-02:
Yes I mean 14.04, sorry for typo.
Nope - I did not run into any errors with dependencies.
I will be installing new Ubuntu server in few days and can try the compilation on different Ubuntu version - which should I use?

dakotabenjamin, 2017-05-02:
16.04 is your best bet. I'll look into 14.04 compilation issues in the meantime.

dakotabenjamin, 2017-05-08:
@johnfox was this a fresh ubuntu installation?

johnfox, 2017-05-09:
Nope the 14.04 was "old" installation. I have successfully compiled on the fresh 16.04 ubuntu installation.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/compilation-fails-on-ubuntu-14-04/394

closed 2018-03-28 by dakotabenjamin