OpenDroneMap/ODM #559: References for ODM

CarlosGrohmann, 2017-05-02:
It would be really nice to have a page with the preferred references for citing ODM in scientific articles. A page in the wiki could also be used to list all papers that use ODM.

dakotabenjamin, 2017-05-02:
Does anyone know how to cite ODM in scientific papers? That is something I would put right in the README.

LucasMM86, 2017-05-02:
Hello Benjamin,
I found something here: http://blog.apastyle.org/apastyle/2015/01/how-to-cite-software-in-apa-style.html

It would be like this:
Reference:
Open Drone Map [Computer software]. (2017) Retrieved from https://github.com/OpenDroneMap/OpenDroneMap

Cite: 
"We used Version 0.2 of Open Drone Map (2017) to complete our work."

But maybe there are better ways to do that, i am not an expert! regards

CarlosGrohmann, 2017-05-02:
Yes, that would be the way to cite the software itself. 

I was thinking on something in the lines of this GRASS page:
https://grasswiki.osgeo.org/wiki/GRASS_Citation_Repository

Are there any scientific papers about ODM?

dakotabenjamin, 2017-05-02:
None published that I am aware of.

dakotabenjamin, 2017-05-02:
This should be linked over at #516

CarlosGrohmann, 2017-05-02:
A good starting point would be a list of references for the parts that make ODM (OpenSfM, libraries, etc). It would be very useful for a paper describing ODM.

CarlosGrohmann, 2017-05-04:
Using Google Scholar, I found some papers using ODM.

I'll make a list, but this one seem to be of special interest:

Comparing and characterizing three-dimensional point clouds derived by structure from motion photogrammetry
by Schwind, Michael, M.Sc., TEXAS A&M UNIVERSITY - CORPUS CHRISTI, 2016, 129 pages; 10247379
http://hdl.handle.net/1969.6/1137

rionlerm, 2017-05-04:
Citing a well-known statistical software package:

'To cite R in publications, use

@Manual{,
  title        = {R: A Language and Environment for Statistical
                  Computing},
  author       = {{R Core Team}},
  organization = {R Foundation for Statistical Computing},
  address      = {Vienna, Austria},
  year         = YEAR,
  url          = {https://www.R-project.org}
}
'

pierotofy, 2020-09-11:
@smathermather what would be the best way to do this? (Add a reference in the README/Docs?)

CarlosGrohmann, 2020-09-11:
Some projects have a section in their website (like Citing XX), but a section in the README works well.

smathermather, 2020-09-12:
We could add to the README and maybe a page at opendronemap.org.

PR #1157 (2020-09-13): Added citation
Adds citation instructions.

Closes #559
closed 2020-09-13 by smathermather