OpenDroneMap/ODM #562: Memory leak? (docker)

CarlosGrohmann, 2017-05-03:
Ok, I understand this might be a common issue, but here it goes:

I just run ODM with a set of 99 images and using all default parameters, except for --force-ccd 6.17 (phantom 4 camera). It took 40min and the results are good.
Memory usage was at about 3GiB for depthmaps and went up to 12GiB for depthmaps cleaning (docker has 20 GiB allocated to it)

After that, I tried to tweak some of the parameters:

```
time docker run -it --rm \
    -v $IMG_DIR:/code/images \
    -v $(pwd)/odm_georeferencing:/code/odm_georeferencing \
    -v $(pwd)/odm_orthophoto:/code/odm_orthophoto \
    -v $(pwd)/odm_texturing:/code/odm_texturing \
    -v $(pwd)/odm_meshing:/code/odm_meshing \
    -v $(pwd)/odm_georeferencing:/code/odm_georeferencing \
    -v $(pwd)/opensfm:/code/opensfm \
    opendronemap/opendronemap \
    --force-ccd 6.17 \
    --resize-to 4000 \
    --min-num-features 10000 \
    --mesh-size 2500000 \
    --mesh-octree-depth 11 \
    --mesh-solver-divide 11 \
    --matcher-neighbors 8 \
    --matcher-distance 0 \
    --opensfm-processes 16 \
    --gcp curralinho.gcps
    --time \
```

This time, I saw the memory usage going from 12GiB for HAHOG to 19.5 for depthmaps.  Then block/IO started to increase from 40-50MB up to 10Gib and then it just stoped (cpu went to 0.5%).

Could it be a memory leak? Any object that should've been garbage-collected but wasn't? 

I'm using the opendronemap/opendronemap docker image from docker hub. This seems to be the latest version, but not the devel one. (is there an opendronemap-dev docker image?)

CarlosGrohmann, 2017-05-04:
OK, I went and installed ODM in a Vagrant machine (with 28 GiB of allocated RAM) and re-run the same set of 99 images, as: 
```
python /odm_app/OpenDroneMap/run.py \
--project-path /odm_data/ \
--images /odm_data/missao_01/ \
--force-ccd 6.17 \
--resize-to 3999 \
--min-num-features 10000 \
--mesh-size 2500000 \
--mesh-octree-depth 11 \
--mesh-solver-divide 11 \
--time \
curralinho_m1_tweaks
```

This time it processed successfully, so the memory thing could be docker-related or version-related, since the vagrant machine is running the latest version of the Master branch. Later I'll try to give docker 28 GiB and see if it works.

pierotofy, 2017-05-04:
@CarlosGrohmann could you detail how you are measuring the memory/disk usage statistics?

CarlosGrohmann, 2017-05-05:
@pierotofy I used a highly scientific method of opening a new terminal window, running docker stats and keeping an eye on it :)

hblanken, 2017-05-15:
Hi @CarlosGrohmann, @pierotofy  - I installed a clean version opendronemap 0.3.1 on ubuntu 16.04 with 15GB RAM yesterday, I ran a 219 image reconstruction, first with --resize-to 2400 then tried with --resize-to 1000 which completed fine on previous releases. 
Now I see the same issue as you did, spikes in memory as soon as depthmaps kickoff and bad memory allocation, is this normal? Sth with opensfm is my suspicion - shall I limit the --opensfm-processes? 
Each of the parallel python processes (I had 17 parallel processes) had about 50k, before depth map computation, then jumped to 3000k each.
ubuntu@xxx:~$ ps --sort -rss -eo rss,pid,command | head
  RSS   PID COMMAND
3070304 13836 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
3063064 13834 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
3063064 13835 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
2178752 13838 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
1976696 6645 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
1465656 13842 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
1465652 13840 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
1465652 13841 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xx/opensfm
551840  520 /lib/systemd/systemd-udevd

ubuntu@xxx:~$ free -m
              total        used        free      shared  buff/cache   available
Mem:          15038       13874

Extract of the commandline log:

2017-05-15 12:13:31,300 Computing depthmap for image DJI_0741.JPG
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 326, in _handle_workers
    pool._maintain_pool()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 230, in _maintain_pool
    self._repopulate_pool()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 223, in _repopulate_pool
    w.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
    self._popen = Popen(self)
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
    self.pid = os.fork()
OSError: [Errno 12] Cannot allocate memory
2017-05-15 12:13:49,865 Computing depthmap for image DJI_0880.JPG
2017-05-15 12:14:07,243 Computing depthmap for image DJI_0839.JPG
2017-05-15 12:14:24,922 Using precomputed raw depthmap DJI_0884.JPG
2017-05-15 12:14:24,922 Using precomputed raw depthmap DJI_0848.JPG
2017-05-15 12:14:24,922 Using precomputed raw depthmap DJI_0756.JPG
2017-05-15 12:14:24,924 Using precomputed raw depthmap DJI_0838.JPG
2017-05-15 12:14:24,925 Using precomputed raw depthmap DJI_0542.JPG
2017-05-15 12:14:24,925 Using precomputed raw depthmap DJI_0877.JPG
2017-05-15 12:14:24,925 Using precomputed raw depthmap DJI_0603.JPG
2017-05-15 12:14:26,252 Computing depthmap for image DJI_0779.JPG
Process PoolWorker-8:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 102, in worker
    task = get()
  File "/usr/lib/python2.7/multiprocessing/queues.py", line 378, in get
    return recv()
MemoryError
2017-05-15 12:14:56,434 Using precomputed raw depthmap DJI_0738.JPG
2017-05-15 12:14:56,436 Using precomputed raw depthmap DJI_0830.JPG
2017-05-15 12:14:56,438 Using precomputed raw depthmap DJI_0787.JPG
2017-05-15 12:14:56,441 Using precomputed raw depthmap DJI_0754.JPG
2017-05-15 12:14:56,444 Using precomputed raw depthmap DJI_0783.JPG
2017-05-15 12:14:56,447 Using precomputed raw depthmap DJI_0554.JPG
2017-05-15 12:14:56,449 Using precomputed raw depthmap DJI_0797.JPG
2017-05-15 12:15:13,862 Computing depthmap for image DJI_0601.JPG

pierotofy, 2017-05-15:
What happens if you limit --opensfm-processes to half the number of physical CPUs?

hblanken, 2017-05-15:
I think we are onto sth. I have a g2.2xlarge	8CPUs, 15GB RAM. I ran it with --opensfm-processes 2
btw, unrelated: - from you experience which box would you recommend for ODM, i.e. best bang for buck and speed? https://aws.amazon.com/ec2/instance-types/
So here is the mem usage:
Start phase:
$ free -m
              total        used        free      shared  buff/cache   available
Mem:          15038        1539       12744
Computing Neighbors - process chews up ~1.5GB
$ free -m
              total        used        free      shared  buff/cache   available
Mem:          15038        2430       11849
Depthmaps with 2 opensfm processes, spikes up to total 10GB mem usage - I also tried --opensfm-processes 4 and it just failed, with 4x~3(for the depthmaps)+2 (for the other processes), but with spikes up to 4GB for the depthmaps and then failing the project.
$ free -m
              total        used        free      shared  buff/cache   available
Mem:          15038        9777        4463
There are 3 heavy processes running:
 RSS   PID COMMAND
4046116 6933 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xxx/opensfm
3085824 6934 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xxx/opensfm
2058312 32097 python /home/ubuntu/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm compute_depthmaps /home/ubuntu/OpenDroneMap/xxx/opensfm

hblanken, 2017-05-15:
@pierotofy Can only complete the job with pmvs-And it completes fast. opensfm cleaning depthmaps kills it all the time. :(

pierotofy, 2017-05-15:
@hblanken could you open an issue in the OpenSfM repository about this? It probably needs to be fixed there.

FarhadG, 2018-06-08:
We're running into the same exact issues with our instance on AWS.

FarhadG, 2018-06-08:
We're experiencing the same exact issue where we exceed the AWS `10GB` allocated Docker memory. Any viable work-arounds until it's resolved?

closed 2020-04-14 by smathermather