OpenDroneMap/ODM #564: Error in orb_SLAM

Ntweat, 2017-05-03:
Hi 

I am getting this error when I try to make the new SLAM module

```
`[ 21%] Performing build step for 'orb_slam2'
make[5]: *** No rule to make target '/usr/lib/libOpenNI.so', needed by '/home/ntweat/opendrone/SuperBuild/src/orb_slam2/lib/libORB_SLAM2.so'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[4]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/orb_slam2.dir/build.make:123: recipe for target 'orb_slam2/stamp/orb_slam2-build' failed
make[2]: *** [orb_slam2/stamp/orb_slam2-build] Error 2
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/orb_slam2.dir/all' failed
make[1]: *** [CMakeFiles/orb_slam2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
ntweat@ntweat-HP-Pavilion-g4-Notebook-PC:~/opendrone/SuperBuild/build$ 
`
```

dakotabenjamin, 2017-05-03:
Are you following the instructions here: https://github.com/OpenDroneMap/OpenDroneMap/wiki/Reconstruction-from-Video?

Ntweat, 2017-05-03:
Yes. 
I am getting the error when I do "make" 
cmake runs fine.

dakotabenjamin, 2018-03-28:
Closing this in favor of #577

closed 2018-03-28 by dakotabenjamin