OpenDroneMap/ODM #566: Spatial distribution of points in CSV and LAS

CarlosGrohmann, 2017-05-04:
Using CloudCompare, I noticed that the spatial distribution of points in the CSV and LAS files are quite different. 

In CSV, points are randomly scattered in the area, while in LAS, they follow narrow strips (see below), and the points are vertically distributed in these strips.

[image]

dakotabenjamin, 2017-05-04:
This is a known bug. I'm trying to find the issue where we've documented this.

pierotofy, 2017-05-04:
Yup, it's documented here: #522

CarlosGrohmann, 2017-05-05:
Ok, since this is a duplicate, I'll close it.
Thanks

closed 2017-05-05 by CarlosGrohmann