OpenDroneMap/ODM #568: Tiles not loaded on Map - with native installation

juvinski, 2017-05-04:
Hi, 

I'm using native installation on ubuntu 15.10 and after generating a map the tiles are not currently displayed.
I'm using as options: min-num-features: 20000, orthophoto-resolution: 40, dem-resolution: 4, verbose: true, dem: true
With dataset: https://www.dropbox.com/s/q7p1bvego9b8efl/flight_nir.zip?dl=0
This are the log of the process:
 (https://github.com/OpenDroneMap/OpenDroneMap/files/977593/log_opendronemap.txt)
This are all results: https://www.dropbox.com/s/2j5j472klcrryx6/project_result.zip?dl=0

Then when I click on view ortophoto at console I see this errors:
WARNING Not Found: /api/projects/19/tasks/46/tiles/1/0/0.png
WARNING "GET /api/projects/19/tasks/46/tiles/1/0/0.png HTTP/1.1" 404 23
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94885/111845.png
WARNING "GET /api/projects/19/tasks/46/tiles/18/94885/111845.png HTTP/1.1" 404 23
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94883/111843.png
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94884/111844.png
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94885/111844.png
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94883/111844.png
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94883/111845.png
WARNING "GET /api/projects/19/tasks/46/tiles/18/94884/111844.png HTTP/1.1" 404 23
WARNING "GET /api/projects/19/tasks/46/tiles/18/94885/111844.png HTTP/1.1" 404 23
WARNING "GET /api/projects/19/tasks/46/tiles/18/94883/111843.png HTTP/1.1" 404 23
WARNING "GET /api/projects/19/tasks/46/tiles/18/94883/111844.png HTTP/1.1" 404 23
WARNING "GET /api/projects/19/tasks/46/tiles/18/94883/111845.png HTTP/1.1" 404 23
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94885/111843.png
WARNING "GET /api/projects/19/tasks/46/tiles/18/94885/111843.png HTTP/1.1" 404 23
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94882/111844.png
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94884/111845.png
WARNING "GET /api/projects/19/tasks/46/tiles/18/94882/111844.png HTTP/1.1" 404 23
WARNING "GET /api/projects/19/tasks/46/tiles/18/94884/111845.png HTTP/1.1" 404 23
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94882/111845.png
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94884/111843.png
WARNING Not Found: /api/projects/19/tasks/46/tiles/18/94882/111843.png
WARNING "GET /api/projects/19/tasks/46/tiles/18/94882/111845.png HTTP/1.1" 404 23
WARNING "GET /api/projects/19/tasks/46/tiles/18/94884/111843.png HTTP/1.1" 404 23
WARNING "GET /api/projects/19/tasks/46/tiles/18/94882/111843.png HTTP/1.1" 404 23

dakotabenjamin, 2017-05-08:
I'll take a look at the output, but you will want to open this issue up at the WebODM repository: https://github.com/OpenDroneMap/WebODM/issues/new it looks like you are running into issues with the viewer.

pierotofy, 2017-05-08:
What's your version of gdal?

`gdalinfo --version`?

You need version >= 2.

juvinski, 2017-05-08:
Hi,
I'm using GDAL 2.1.3, released 2017/20/01
In true, WebODM is showing file not found - http error 404 - or the WebODM is pointing to wrong place or tiles are not generated - I cannot find the tiles WebODM is looking for - the file names are not present in the folders.

pierotofy, 2017-05-08:
Yes, as you can read from the log file:

```
ERROR 6: EPSG PCS/GCS code 900913 not found in EPSG support files.  Is this a valid
EPSG coordinate system?
ERROR 6: No translation for an empty SRS to PROJ.4 format is known.
ERROR 6: No translation for an empty SRS to PROJ.4 format is known.
```

900913 is not a valid EPSG code (https://gis.stackexchange.com/questions/40538/what-is-the-difference-between-epsg900913-and-epsg3857).

You probably have an old copy of gdal2tiles.py, which is not handling the EPSG code correctly. Either upgrade gdal2tiles.py to a version >=2.1.1, or manually patch it via:

```
--- /usr/bin/gdal2tiles.py	2014-04-05 14:21:17.000000000 +0000
+++ gdal2tiles.py	2016-11-08 19:33:42.162188732 +0000
@@ -783,7 +783,7 @@
         self.out_srs = osr.SpatialReference()
 
         if self.options.profile == 'mercator':
-            self.out_srs.ImportFromEPSG(900913)
+            self.out_srs.ImportFromEPSG(3857)
         elif self.options.profile == 'geodetic':
             self.out_srs.ImportFromEPSG(4326)
         else:
```

juvinski, 2017-05-09:
Hi @pierotofy 
you was right, the problem was in gdal2tiles.py.
I updated gdal to 2.1.3 and now is working.

closed 2017-05-09 by juvinski