OpenDroneMap/ODM #569: PC freeze running v0.3 on image set

rionlerm, 2017-05-05:
Hi all,
Ubuntu 14.04
PC with 12 Gig RAM
300 photos
v.0.3

Runs with previous versions on same photo set delivered results.
Currently script freezes together with PC at <...cleaning depthmap on image...>

Suggestions?

pierotofy, 2017-05-06:
Hi @rionlerm, we've had quite a few people reporting issues with the depthmap calculation process stalling. The best workaround at the moment is to use the `--use-pmvs` flag during processing.

pierotofy, 2017-05-06:
Also if you could provide us a copy of the dataset to reproduce the issue it would help us narrow down the root cause of this.

dakotabenjamin, 2017-05-08:
Maybe we should just default nproc-1 for opensfm-processes

pierotofy, 2017-05-08:
Could be a good temporary workaround. Better slow than freezing.

closed 2018-03-28 by dakotabenjamin