OpenDroneMap/ODM #574: Hanging on 'computing neighbours' process

everydayduffy, 2017-05-12:
Hello,

I'm running ODM with Docker on Windows 10 64bit. I have 16GB RAM. 

I ran the following command:

`docker run -it --rm -v C:/Software/OpenDroneMap/images:/code/images -v C:/Software/OpenDroneMap/odm_orthophoto:/code/odm_orthophoto -v C:/Software/OpenDroneMap/odm_texturing:/code/odm_texturing my_odm_image`

And processing got as far as is shown here:

[image]

This has been hanging on this stage for approximately 12 hours.

My `images` folder contains 207 16megapixel JPEG images. I was wondering if this is a realistic time for ODM to spend on the computing neighbours process. 

Any help much appreciated.

Thank you.

everydayduffy, 2017-05-12:
This was an issue with the default docker settings not allocating enough RAM. Solved.

closed 2017-05-12 by everydayduffy

everydayduffy, 2017-05-12:
Updated Wiki in case anyone else found this is an issue.

everydayduffy, 2017-05-12:
@dakotabenjamin I'm still having issues with v0.3 in general - hanging at other stages such as depth map computation though. Reverting to --use-pmvs has solved it. Let me know if you'd like me to open a new issue.

dakotabenjamin, 2017-05-12:
What's your processor? The depth-map issue is a known bug, the current workaround is to use pmvs or run with `--opensfm-processes n` where in is $(nproc) - 1

everydayduffy, 2017-05-12:
Intel Core i7-4800MQ 2.7GhZ. Been using pmvs without issues bu haven't tried the processes fix. I will give that a go.

everydayduffy, 2017-05-15:
@dakotabenjamin - do you have an equivalent `nproc` command for windows please? I'm guessing my processor can handle 8 processes. To be safe I set `--opensfm-processes 3` but it still hung for over 24 hours.

dakotabenjamin, 2017-05-16:
I think the best thing would just use the number you think works. If you're happy with the output from pmvs I would keep working with that until we get depth mapping under better management for lower-end machines.