OpenDroneMap/ODM #575: "The killer feature" [Request]

mfr84, 2017-05-14:
To date, there is no photogrammetric software (in the entire world) that can pull out "measurements" without powerful hardware and without long processing time.

OpenDroneMap has already opened the photogrammetric technology at all... it would be great if it will free it also from having a 2k$ hardware and 4 hours of time.

My dream is to go on the field, fly my drone, download data on a low-end I3 laptop with 4 GB RAM and have distances, heights and countours in a snap with "environmental engineering approximation"...

pierotofy, 2017-05-22:
Mm, 4GB RAM and results "in a snap" might be difficult to achieve 😄 

Is this a request for a specific feature? We have the ability to measure distances and heights in WebODM. Contours can be generated separately at the moment in GRASS or QGIS.

dakotabenjamin, 2017-05-23:
RAM is cheap and easier to replace on a laptop than a CPU, I would go with 16 if you can. I like @pierotofy's suggestion better though, it would be easier to use WebODM- you can have a much more capable machine and you only have to concern yourself with data upload.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/the-killer-feature-request/371

closed 2018-03-28 by dakotabenjamin