OpenDroneMap/ODM #577: Cannot run video reconstruction

syehorov, 2017-05-15:
Hi. Got a problem with running video reconstruction for current GIT snapshot. Using proposed castle video for tests. ODM is running inside Docker, image made using Dockerfile:

```
FROM ubuntu
ENV DEBIAN_FRONTEND noninteractive
RUN apt-get update && apt-get remove libdc1394-22-dev \
     && apt-get install -y --install-recommends sudo build-essential cmake \
     python-pip libgdal-dev libgeotiff-dev pkg-config libgtk2.0-dev \
     libavcodec-dev libavformat-dev libswscale-dev python-dev python-numpy \
     libtbb2 libtbb-dev libjpeg-dev libpng-dev libtiff-dev libjasper-dev \
     libflann-dev libproj-dev libxext-dev liblapack-dev libeigen3-dev \
     libvtk5-dev python-networkx libgoogle-glog-dev libsuitesparse-dev \
     libboost-filesystem-dev libboost-iostreams-dev libboost-regex-dev \
     libboost-python-dev libboost-date-time-dev libboost-thread-dev \
     python-empy python-nose python-pyside python-pyexiv2 python-scipy jhead \
     liblas-bin libglew-dev python-catkin-pkg catkin libgmp-dev libgmp3-dev \
     libmpfr-dev libmpfrc++-dev vim-nox \
     && apt-get autoremove && apt-get clean
RUN useradd -m -U odm
RUN mkdir /code
WORKDIR /code
RUN git clone https://github.com/OpenDroneMap/OpenDroneMap.git .
RUN git submodule init && git submodule update
RUN wget -c http://launchpadlibrarian.net/165141962/libeigen3-dev_3.2.0-8_all.deb && dpkg -i libeigen3-dev_3.2.0-8_all.deb && rm -f libeigen3-dev_3.2.0-8_all.deb
RUN cd SuperBuild && cmake -DODM_BUILD_SLAM=ON . && make
RUN bash ./configure.sh install && \
    cd build && cmake -DODM_BULID_SLAM=ON .. && make && cd .. && \
    chown -R odm:odm /code
USER odm
ENV PYTHONPATH=${PYTHONPATH}:/code/SuperBuild/install/lib/python2.7/dist-packages:/code/SuperBuild/src/opensfm \
    LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/code/SuperBuild/install/lib
VOLUME ["/images"]
ENTRYPOINT ["python", "/code/run.py", "--project-path", "/images"]

```

It works fine for images, but when trying to do video I get next:

```
docker run -t -i --entrypoint=/bin/bash -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/prapor/Downloads:/code/video opendronemap
odm@53e1ad7fbae3:/code$ RUNPATH="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
odm@53e1ad7fbae3:/code$ export PYTHONPATH=$RUNPATH/SuperBuild/install/lib/python2.7/dist-packages:$RUNPATH/SuperBuild/src/opensfm:$PYTHONPATH
odm@53e1ad7fbae3:/code$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$RUNPATH/SuperBuild/install/lib
odm@53e1ad7fbae3:/code$ python modules/odm_slam/src/calibrate_video.py --visual 
./video/output/test1/chessboard.mp4 
kept    current chessboard found
34      1107    True              
RMS: 0.881957891046

# Camera calibration and distortion parameters (OpenCV)
Camera.fx: 1512.91332401
Camera.fy: 1512.04223185
Camera.cx: 956.585155225
Camera.cy: 527.321715394

Camera.k1: 0.140581949184
Camera.k2: -0.292250537695
Camera.p1: 0.000188785464717
Camera.p2: 0.000611510377372
Camera.k3: 0.181424769625
odm@53e1ad7fbae3:/code$ python run.py --project-path ./video/output --video castle_video.mp4 --slam-config config.yaml --resize-to 1280 test1
[INFO]    Initializing OpenDroneMap app - Mon May 15 13:16:24  2017
Traceback (most recent call last):
  File "run.py", line 39, in <module>
    app = ODMApp(args=args)
  File "/code/scripts/odm_app.py", line 26, in __init__
    ecto.BlackBox.__init__(self, *args, **kwargs)
  File "/code/SuperBuild/install/lib/python2.7/dist-packages/ecto/blackbox.py", line 259, in __init__
    self.__connect()
  File "/code/SuperBuild/install/lib/python2.7/dist-packages/ecto/blackbox.py", line 284, in __connect
    raise BlackBoxError('Got error "%s" when calling connections on "%s"' % (str(e), str(self)))
ecto.blackbox.BlackBoxError: Got error "global name '_p' is not defined" when calling connections on "<scripts.odm_app.ODMApp object at 0x7f2689801c50>"
```

config.yaml is in same folder as chessboard.mp4 and castle_video.mp4, chessboard output is checked to be same as in config. Can somebody say, what I`m doing wrong?

cappelletto, 2017-11-22:
Hello,
I've the same error. In my case, I'm running on Ubuntu 16.04 64 bits, and ODM compiled with ORB_SLAM after fixing some issues with Eigen 3.3 (downgraded to 3.2)
As in your case, the input video and yaml config file are placed in the  /project/path folder, yet I receive the same error message.

Any hints?

dakotabenjamin, 2017-11-28:
I am looking for a champion to get SLAM working again

jeffreyzou, 2017-12-13:
The problem is a bug at scripts\ODM_app.py, line 100,   it should be  return self.slam_connections(p) not '_p'

smathermather, 2020-04-14:
We have removed the video portions of ODM I think... confirming with @pierotofy and closing in the meantime.

closed 2020-04-14 by smathermather

pierotofy, 2020-04-14:
Confirm, SLAM support has been removed (would still be cool to have video support, but without a contribution this is likely not going to get done).