OpenDroneMap/ODM #578: Improve default values for DEM

pierotofy, 2017-05-15:
I thought I'd report my findings here. :)

Whenever I process a dataset, I seem to need to tweak the `--dem-sample-radius`, `--dem-resolution` and `--dem-radius`. The default values produce spotted results.

Input: https://github.com/pierotofy/drone_dataset_sheffield_park_1 (default options plus `--dem`)

Output:

[image]

With `--dem-radius 2`:

[image]

(Resolution is quite low)

With `--dem-radius 2 --dem-resolution 0.2`:

[image]

(Resolution is good, almost the same as the orthophoto, but interpolated values are "bubbly" looking, see the image below)

[image]

With `--dem-sample-radius 0.25 --dem-resolution 0.2 --dem-radius 0.5`:

[image]

Still some holes, but resolution is good and no "bubbly" values. It also took 16 minutes to process the PDAL pipeline, if speed is a concern, perhaps adding the "approximate" option to pmf could speed things up: https://www.pdal.io/stages/filters.pmf.html

With `--dem-sample-radius 0.25 --dem-resolution 0.2 --dem-radius 1`:

[image]

Mostly OK, still some holes, took 17 minutes. I suspect increasing the max_window_size in pmf would interpolate the nodata cells, but would also add significant overhead.

In general I think the current defaults are undersampling and the resolution might be too low. There should also be more hole filling perhaps? Just suggestions. 👍 love the new DEM functionality.

smathermather, 2017-05-15:
I really think we just need better interpolators in PDAL. Great interpolation isn't needed for lidar, but is critical for photogrammetrically derived point clouds.

pierotofy, 2017-05-22:
Relevant: https://github.com/OpenDroneMap/node-OpenDroneMap/issues/15

dakotabenjamin, 2018-03-28:
This is irrelevant now, correct? Please reopen if I'm wrong.

closed 2018-03-28 by dakotabenjamin