OpenDroneMap/ODM #583: [Enhancement] Add StatisticalOutlierRemoval to odm_meshing module

pierotofy, 2017-05-18:
Many datasets end up having small artifacts lying around both the textured mesh and orthophoto. Many of these small artifacts could be prevented by removing outliers prior to mesh building.

See for example:

[image]

We've already have a `--mesh-remove-outliers` for the 2.5D mesh, we could probably use it for the 3D mesh as well.

dakotabenjamin, 2017-05-23:
The question is whether there's a functional equivalent, or have to be added to the module manually. Any C++ devs want to take up the challenge?

pierotofy, 2017-05-23:
I could probably add it to the meshing module: http://docs.pointclouds.org/trunk/classpcl_1_1_statistical_outlier_removal.html

mribbons, 2017-05-25:
👍  this is a great suggestion.

I'm not a c++'er unfortunately, but I could assist with testing and debug once some code is written.

osmanmesutozcan, 2017-06-07:
I can help with this, if there is no one working on it.

pierotofy, 2017-06-07:
I haven't started working on this, so go ahead, it would be awesome to have this feature!

hy9be, 2017-08-10:
It would be great to have this!

smathermather, 2020-04-14:
Done since 2018.

closed 2020-04-14 by smathermather