OpenDroneMap/ODM #586: Only orthophoto?

Davidlapalma, 2017-05-23:
Is there any posibility to run only orthophoto image without the program run the 3D mesh and others operations that don't be useful? And it's not possible what is the structure of PMVS (all the files that are composed) because this is the most heaviest computing processing part and i would like reduce its processing time?

dakotabenjamin, 2017-05-23:
Each step builds off the last, so you must run the whole process. To be honest it's been a while since I've used the CMVS/PMVS part of the toolchain but I'll try to help you understand it. You can read more about optimizing PMVS here: http://www.di.ens.fr/cmvs/ 

Here's the file structure for PMVS:
```
|-- models/                                # The important part
    |-- option-0000.ply               # This one is the output
|-- txt/                                       # Info about each image used in the reconstruction
    |-- 00000000.txt
    |-- ...etc
|-- visualize/                            # The images used for building the reconstruction
    |-- 00000000.jpg
    |-- ...etc
|-- bundle.rd.out                     # the bundle file (from sfm)
|-- centers-0000.ply                # Camera positions used in reconstruction
|-- centers-all.ply                    # All cameras
|-- option-0000                      # the generated params for pmvs
|-- pmvs.sh                             # The bash file to run pmvs with option-0000 as params
|-- pmvs_options.txt               # produced by ODM to generate option-0000
|-- ske.dat                               # some info file
|-- vis.dat                                # some info file
```

Davidlapalma, 2017-05-23:
Thank you very much Dakota I hope that support me to find a solution

kikislater, 2017-05-23:
I think you want just a photo merge like microsoft ice or a panorama like software. 
If we have imu (aka orientation) it could be usefull especially for multispectral cameras where we need only photostitch ...

dakotabenjamin, 2018-03-28:
We've implemented a `--fast-orthophoto` flag which skips some of the heavy computation  steps and should speed up processing.

closed 2018-03-28 by dakotabenjamin