OpenDroneMap/ODM #588: Eigen problems

patrickelectric, 2017-05-24:
I tried to compile it and was not able.
- eigen 3.3.3-1 
```
/mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp: In member function ‘void OdmOrthoPhoto::createOrthoPhoto()’:
/mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:445:58: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
         mesh.tex_coordinates.push_back(textureCoordinates);
                                                          ^
In file included from /usr/include/c++/6.3.1/vector:64:0,
                 from /usr/include/pcl-1.8/pcl/PCLPointCloud2.h:9,
                 from /usr/include/pcl-1.8/pcl/TextureMesh.h:44,
                 from /usr/include/pcl-1.8/pcl/io/obj_io.h:40,
                 from /mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.hpp:9,
                 from /mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:7:
/usr/include/c++/6.3.1/bits/stl_vector.h:914:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(const value_type& __x)
       ^~~~~~~~~
/usr/include/c++/6.3.1/bits/stl_vector.h:914:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘const value_type& {aka const std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >&}’
/usr/include/c++/6.3.1/bits/stl_vector.h:932:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(std::vector<_Tp, _Alloc>::value_type&&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(value_type&& __x)
       ^~~~~~~~~
/usr/include/c++/6.3.1/bits/stl_vector.h:932:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::value_type&& {aka std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >&&}’
/mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp: In member function ‘bool OdmOrthoPhoto::loadObjFile(std::__cxx11::string, pcl::TextureMesh&)’:
/mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:1216:60: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
                 mesh.tex_coordinates.push_back (coordinates);
                                                            ^
In file included from /usr/include/c++/6.3.1/vector:64:0,
                 from /usr/include/pcl-1.8/pcl/PCLPointCloud2.h:9,
                 from /usr/include/pcl-1.8/pcl/TextureMesh.h:44,
                 from /usr/include/pcl-1.8/pcl/io/obj_io.h:40,
                 from /mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.hpp:9,
                 from /mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:7:
/usr/include/c++/6.3.1/bits/stl_vector.h:914:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(const value_type& __x)
       ^~~~~~~~~
/usr/include/c++/6.3.1/bits/stl_vector.h:914:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘const value_type& {aka const std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >&}’
/usr/include/c++/6.3.1/bits/stl_vector.h:932:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(std::vector<_Tp, _Alloc>::value_type&&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(value_type&& __x)
       ^~~~~~~~~
/usr/include/c++/6.3.1/bits/stl_vector.h:932:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::value_type&& {aka std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >&&}’
/mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:1264:54: error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’
         mesh.tex_coordinates.push_back(texcoordinates);
                                                      ^
In file included from /usr/include/c++/6.3.1/vector:64:0,
                 from /usr/include/pcl-1.8/pcl/PCLPointCloud2.h:9,
                 from /usr/include/pcl-1.8/pcl/TextureMesh.h:44,
                 from /usr/include/pcl-1.8/pcl/io/obj_io.h:40,
                 from /mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.hpp:9,
                 from /mnt/cd7c8fe2-0fea-413c-be99-fcf387c77815/git/horus/OpenDroneMap/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp:7:
/usr/include/c++/6.3.1/bits/stl_vector.h:914:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(const value_type& __x)
       ^~~~~~~~~
/usr/include/c++/6.3.1/bits/stl_vector.h:914:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘const value_type& {aka const std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >&}’
/usr/include/c++/6.3.1/bits/stl_vector.h:932:7: note: candidate: void std::vector<_Tp, _Alloc>::push_back(std::vector<_Tp, _Alloc>::value_type&&) [with _Tp = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >; std::vector<_Tp, _Alloc>::value_type = std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > >]
       push_back(value_type&& __x)
       ^~~~~~~~~
/usr/include/c++/6.3.1/bits/stl_vector.h:932:7: note:   no known conversion for argument 1 from ‘std::vector<Eigen::Matrix<float, 2, 1> >’ to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >:: [... cut ...] en::Matrix<float, 2, 1> > >; _Alloc = std::allocator<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >]
       operator=(initializer_list<value_type> __l)
       ^~~~~~~~
/usr/include/c++/6.3.1/bits/stl_vector.h:471:7: note:   no known conversion for argument 1 from ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1> > >’ to ‘std::initializer_list<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >’
```

XiaoshuiHuang, 2017-07-21:
have you solved your problem?

dakotabenjamin, 2017-07-21:
Hi are you using 14.04? The problem and workaround is here: #613 I'm going to close this as a duplicate. Please reopen if your problem is different or remains unsolved.

patrickelectric, 2017-07-21:
@dakotabenjamin I am using archLinux. Will try again with this solution later.

dakotabenjamin, 2017-07-21:
@patrickelectric you may have a different problem. Currently ODM is only natively supported by Ubuntu by us, although I know its been made to work with Debian. We currently don't have any support for archLinux. You may be able to get it working by installing all the dependencies in the configure.sh file manually and creating links to ODM, but I don't know how much help we can be in that endeavor. 

Does Docker work with archLinux? I would recommend trying that using the docker intructions in the readme.

patrickelectric, 2017-07-21:
@dakotabenjamin I will try here now.
I will update configure.sh to work with arch.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/eigen-problems/392

closed 2018-03-28 by dakotabenjamin

MerlotQ, 2018-11-20:
It contains a pcl definition problem. PCL 1.8.0 has fixed this.